rusefi/rusefi #5595: Problem with downloading simulator

krisyovchevv, 2023-10-05:
Hello, I tried to download the simulator for TunerStudio, but after clicking on the link to download the zip file, I get a tab that shows an error and gives me two buttons:
1- Main page
2- New wiki
Is there a way to download the simulator or not?
[image]

rusefillc, 2023-10-05:
@krisyovchevv please elaborate on how exactly are you trying to download the simulator? As is I have no idea what are you clicking where meaning I would not be able to fix.

krisyovchevv, 2023-10-05:
I go to the github page for the guide on how to download and use the simulator: https://github.com/rusefi/rusefi/wiki/Virtual-simulator
and click on the "here" button, where the zip file with the files for the simulator should be downloaded:
[image]

Whereupon after pressing the "here" button, I get a page saying "Not found", and below two options:
[image]

rusefillc, 2023-10-05:
Fixed link at https://github.com/rusefi/rusefi/wiki/Virtual-simulator

closed 2023-10-05 by rusefillc

krisyovchevv, 2023-10-05:
Thank you <333