rusefi/rusefi #5596: Retarding exhaust VVT which drives DI pump causes issues

ElDominio, 2023-10-05:
**Describe the bug**
An exhaust VVT that drives to retard, and the base pump angle is 10 degrees, any time you target less than -10 degrees of VVT retard, GDI pump angle breaks.

**Hardware**
4chan
PB Adapter
GDi4

**To Reproduce**
Steps to reproduce the behavior:
1. set up VVT as normal
2. control exhaust VVT
3. set target angle to under -10 (in this case)
4. watch gdi pump angle go crazy

**Expected behavior**
GDi angle moves consistently

**Screenshots**
No screenshots but log at
https://rusefi.com/online/view.php?log=1297

check GDi pump angle and exhaust VVT angle (GDi pump is driven by exhaust cam)

rusefillc, 2023-10-06:
problem: negative enginePhase argument would cause AngleBasedEvent to get forever stuck in TriggerSchedule

on the other hand all lobes share the same instance of AngleBasedEvent and we would repurpose that invalid AngleBasedEvent with a more valid one

lastly potentially we have yet another issue - could it be that we keep scheduling different phaseEngle for each lobe while never actually having a chance to fire that event?

ElDominio, 2023-10-06:
[image]

At 0 degrees of exhaust VVT, next GDi pump angle target looks to be reasonable.

[image]

At -20, looks confused, target is mostly flat?

[image]

At -15, angle changes, but it does not follow the recognizable pattern

ElDominio, 2023-10-10:
https://rusefi.com/online/view.php?log=1305

Still the same issue, no general change

rusefillc, 2023-10-11:
[image]

rusefillc, 2023-10-25:
Depends on #5640

rusefillc, 2024-03-18:
we believe -50 lobe peak to be more correct? better? red kia value

[image]

rusefillc, 2024-03-18:
way to make it bad

[image]

commit 19dde2fc (2023-10-06): High Pressure Fuel Pump scheduling goes into negative range fix #5596