rusefi/rusefi #5608: Flat shift brain dump

ElDominio, 2023-10-14:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I am shifting between gears (in any kind of transmission), and I need to let go of the accelerator to reduce torque to the transmission/rear wheels.

**Would we need new configuration fields?**
There would have to be a new menu under "Advanced" called "Shift Torque Reduction (Flat Shift)", containing **at least** the following settings:
Torque Reduction Trigger Pin- pin that activates the reduction/cut for shifting; **sometimes shared with the Launch Control pin.**
Torque Reduction Time - For how long after the pin has been triggered will the cut/reduction stay active (usually about 200ms). After that, even if the pin is still triggered, torque is re-introduced
Torque Reduction Arming RPM - Since this pin is usually shared with launch control, most people have an RPM where behavior under that is Launch Control, over that is Flat Shift/Torque Reduction
Torque Reduction Arming APP - Pedal position to realize that we need to reduce torque when the trigger pin is uuuh triggered
Torque Reduction Ignition Retard - How many degrees of timing advance will be reduced during the Torque Reduction Time.
Torque Reduction Ignition Cut - How many % of ignition events will be cut (similar to setSparkHardSkipRatio in Lua). 0-100%

This can later be complicated with much more things, but for now this would work good

**Describe the solution you'd like**
I want torque to be reduced by using timing controls when I trigger a pin, for a set amount of time and over a certain RPM and TPS, to assist in transmission shifting.

**Describe alternatives you've considered**
Writing this in Lua, but time constraints and possible memory constraints

ElDominio, 2024-02-23:
I did this with Lua, https://rusefi.com/forum/viewtopic.php?f=5&t=2670

ElDominio, 2024-08-08:
> [@ElDominio](https://github.com/ElDominio), above you listed the following settings on `Shift Torque Reduction (Flat Shift)` dialog:
> 
> * `Torque Reduction Trigger Pin`
> * `Torque Reduction Time`
> * `Torque Reduction Arming RPM`
> * `Torque Reduction Arming APP`
> * `Torque Reduction Ignition Retard`
> * `Torque Reduction Ignition Cut`.
> 
> Could you please map them on variables in your Lua scripts?
> 
> For example am I correct that `Torque Reduction Time` corresponds to `flatShiftTotalTimeMs`?

Sorry for the late reply

you are correct on torque reduction time

my script has separate times to change the amount of torque reduction over time, providing three different windows of torque reduction, where first we are entering a cut (so we start cutting ignition events and timing but not as much as the main phase), then a main phase (where we apply most of this timing cut/retard), and an exit phase that allows us to ramp back up to normal timing/no cut).

in line 9, 
(RPMread) > 2000, RPMread would be Torque Reduction Arming RPM, 
clutchState would be Torque Reduction Trigger Pin, 
driverD would be Torque Reduction Arming APP`

then inside our first if at line 15
setSparkHardSkipRatio would be Torque Reduction Ignition Cut
setTimingAdd would be  Torque Reduction Ignition Retard

ElDominio, 2024-09-19:
Any progress on this?

kifir23917, 2024-09-19:
I'm going to return to this ticket tomorrow after I finish my work on console.

kifir23917, 2024-09-24:
> Torque Reduction Arming APP - Pedal position to realize that we need to reduce torque when the trigger pin is uuuh triggered

@ElDominio, does any existing functionality already use `pedal position`? Do we already have an existing setting in TunerStudio for pedal position to use as sample?

ElDominio, 2024-09-24:
> > Torque Reduction Arming APP - Pedal position to realize that we need to reduce torque when the trigger pin is uuuh triggered
> 
> @ElDominio, does any existing functionality already use `pedal position`? Do we already have an existing setting in TunerStudio for pedal position to use as a sample?

It should be the same value used in TPS Threshold for Launch Control

kifir23917, 2024-09-27:
> Torque Reduction Trigger Pin- pin that activates the reduction/cut for shifting; sometimes shared with the Launch Control pin.

@ElDominio, for `Launch Button` pin we have settings `Launch Button inverted` and `Launch Button Mode`:

[image]

Let's consider case, when we reuse `Launch Button` pin for activation the reduction/cut for shifting. Is my understanding correct that in this case for activation the reduction/cut for shifting we should take into account `Launch Button inverted` and `Launch Button Mode` settings as well?

Now let's consider case when we use for activation the reduction/cut for shifting the pin different from `Launch Button`. Do we need to add `Torque Reduction Button inverted` and `Torque Reduction Button Mode` settings for this case as we do for `Launch Button`?

ElDominio, 2024-09-27:
> > Torque Reduction Trigger Pin- pin that activates the reduction/cut for shifting; sometimes shared with the Launch Control pin.
> 
> @ElDominio, for `Launch Button` pin we have settings `Launch Button inverted` and `Launch Button Mode`:
> 
> ![Image](https://private-user-images.githubusercontent.com/159958110/371551161-c02cfa9f-f1f5-42c4-a537-1116f700c972.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc0NzU0MzAsIm5iZiI6MTcyNzQ3NTEzMCwicGF0aCI6Ii8xNTk5NTgxMTAvMzcxNTUxMTYxLWMwMmNmYTlmLWYxZjUtNDJjNC1hNTM3LTExMTZmNzAwYzk3Mi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyN1QyMjEyMTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03NTgwNzhkYzEyNDQ4YzI1OTY5YmE2YWZiZWQ2ODA2OTdmMjgyMTEzOTRiZDEzOGRmMDE3NTU1ZDllMTU4YzhkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Smx2QYQjkLOnakBjoBCnjAJpQz8BFf-xXfQ1Tcp7r1g)
> 
> Let's consider case, when we reuse `Launch Button` pin for activation the reduction/cut for shifting. Is my understanding correct that in this case for activation the reduction/cut for shifting we should take into account `Launch Button inverted` and `Launch Button Mode` settings as well?
> 
> Now let's consider case when we use for activation the reduction/cut for shifting the pin different from `Launch Button`. Do we need to add `Torque Reduction Button inverted` and `Torque Reduction Button Mode` settings for this case as we do for `Launch Button`?

if we're reusing the Launch Button input, it should be triggered with the same conditions as launch (meaning it engages exactly the same as Launch in terms of pin inversion and mode, but when it is triggered OVER the Launch RPM (meaning with Torque Reduction DISABLED, we would engage Launch control, but with Torque Reduction ENABLED, we would then go into torque reduction control)

Now when using a different pin, we should have configurable Inverted and Mode dialogs

kifir23917, 2024-09-28:
I would suggest for this functionality UI with `Activation Mode` as we already have for `Launch Control` functionality:

[image]

ElDominio, 2024-09-28:
This is a seperate menu from Launch, right? If so, it looks great

kifir23917, 2024-10-03:
> Torque Reduction Arming RPM - Since this pin is usually shared with launch control, most people have an RPM where behavior under that is Launch Control, over that is Flat Shift/Torque Reduction

@ElDominio, does it mean that `Launch Control` module should check `Torque Reduction Arming RPM(rpm)` setting if `Enable Shift Torque Reduction` setting is `true`? And in this case it should perform PreLaunch/Launch adjustments only if RPM is below this setting?

ElDominio, 2024-10-03:
Correct, over `Torque Reduction Arming RPM(rpm)` it should perform Torque Reduction stuff, and below it should do what you said

kifir23917, 2024-10-03:
> Torque Reduction Arming APP - Pedal position to realize that we need to reduce torque when the trigger pin is uuuh triggered
> It should be the same value used in TPS Threshold for Launch Control

@ElDominio, is my understanding correct that to activate torque reduction we need `DriverThrottleIntent` sensor to return a value ABOVE `Torque Reduction Arming APP` setting from the screenshot above?

ElDominio, 2024-10-03:
Correct

kifir23917, 2024-10-05:
@ElDominio, Flat Shift functionality is ready for manual testing. I'm still working on automated tests for this functionality, but you already can touch it.

ElDominio, 2024-10-28:
Just tested this @kifir23917 , seems to work quite well! Only need the Torque Reduction Time to be increased to 2000ms

kifir23917, 2024-11-02:
@ElDominio, I've implemented functionality you asked for in previous comment:

[image]

Could you please test it?

rusefillc, 2024-11-21:
status update: @kifir23917 plans more test coverage

ElDominio, 2024-12-03:
Tested yesterday in my specific car usage; works great!

closed 2024-12-03 by rusefillc