rusefi/rusefi #5609: In Launch Control, REMOVE Hard Cut RPM Adder

ElDominio, 2023-10-14:
**Describe the bug**
The word "adder" in Launch settings can be an adder or a substractor; what?

**Hardware**
All, firmware issue

**To Reproduce**
Steps to reproduce the behavior:
1. Go To Advanced -> Launch Control
2. Check Ignition Retard Adder RPM and Hard Cut Adder RPM

**Expected behavior**
Both work in the same direction is the expected behavior.

**Current Behavior**
Launch RPM = x
Ignition Retard Adder is the range of RPM **FROM** the Launch  RPM that the ignition retard is applied (so X - Ignition Retard Adder = the RPM where Ignition Retard begins)  , while Hard Cut Adder RPM is an ADDITIONAL RPM FROM X, so X + Hard Cut Adder RPM = new real Launch RPM.

It should be consistent, and I argue it should be in the negative direction.
"Hard Cut Adder RPM" should be removed, and Launch RPM IS the hard cut, then the ignition retard adder manages ignition retard in the window specified which is already there.

kifir23917, 2024-05-16:
[image]
@ElDominio, this is a screenshot of current `Launch Control` dialog. Is my understanding correct that the issue is already fixed? Or could you please reword the problem using controls from this screenshot?

kifir23917, 2024-05-20:
@ElDominio, currently I do not see neither "Ignition Retard Adder RPM" nor "Hard Cut Adder RPM" fields on "Launch Control" dialog:
[image]
Does it mean that this ticket is already fixed? Or could you please reword the problem using new UI?

closed 2024-05-20 by rusefillc