rusefi/rusefi #5610: In Launch Control, add RPM Hysterisis

ElDominio, 2023-10-14:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I can't get the expected launch control behavior from my ECU.

**Would we need new configuration fields?**
Launch RPM Hysteresis  - Once the Launch RPM has been hit (and the chosen cut method has been applied) , how many RPMs does the engine need to fall to reactive the previously cut things (be it fuel or ignition)?

**Describe the solution you'd like**
If I set my launch RPM to 4500rpm, and my RPM Hysteresis to 200rpm and my ignition cut to ON, my engine should disable ignition at 4500rpm, and then come back to life at 4300RPM.

rusefillc, 2024-05-14:
whatever @ElDominio talks about above would replace current "window" setting which is a more limited variation of what is desired

@kifir23917 this one here is one of the good tickets to begin with

ElDominio, 2024-05-14:
[image]

As a first actionable item, i think we need to fix this; current interaction between "Launch RPM" and " Hard Cut RPM Adder"  is very strange

kifir23917, 2024-05-20:
@ElDominio, currently on `Launch Control` dialog we already have `Launch Control Window(RPM)` field in addition to `Launch RPM(rpm)` field. These fields define the diapason starting from `Launch RPM(rpm)` - `Launch Control Window(RPM)` to `Launch RPM(rpm)` and we are going to use this diapason to determine in our code `RPM Launch Control` state:
1. NOT_MET (rpm sensor returned value less then `Launch RPM(rpm)` - `Launch Control Window(RPM)`)
2. PRE_LAUNCH (rpm sensor returned value from  `Launch RPM(rpm)` - `Launch Control Window(RPM)` to `Launch RPM(rpm)`)
3. LAUNCH (rpm sensor returned value not less then `Launch RPM(rpm)`).
Would you have an objection if I reuse this Launch Control Window for RPM Hysterisis functionality instead of adding new "Launch RPM Hysteresis" field.
[image]

ElDominio, 2024-05-20:
I think rewording "window" to "hysteresis" is an acceptable change; this ticket was made before those were the names

Change "Launch Control Window" to "Launch RPM Hysteresis"

kifir23917, 2024-05-20:
I would suggest to leave "window" term because we already have pull request that suggest using of **PRE_LAUNCH** state not only for ignition RPM hysteresis, but also for advance angle calculation as well.

ElDominio, 2024-05-20:
that's fine then, leave it as is

kifir23917, 2024-05-21:
> and my ignition cut to ON
I'm implementing hysteresis functionality it a way affecting both `Ignition Cut` and `Fuel Cut` functionality are affected. I mean that if both `Ignition Cut RPM` and `Fuel Cut` are ON and sensor value goes down in `Launch Control Window(RPM)` after it exceeded `Launch RPM(rpm)`, than both ignition and fuel will remain cut. Is it expected behaviour?

kifir23917, 2024-05-29:
From Telegram:

> The hysteresis functionality no longer fully disables spark correct?

> Correct.
> From code I see that we disable spark only when we are in "launch" state and rpm exceed ["Launch RPM(rpm)"]

> That's perfect

closed 2024-05-29 by rusefillc

kifir23917, 2024-06-13:
Hysteresis functionality is [rolled back](https://github.com/rusefi/rusefi/pull/6616) because of https://github.com/rusefi/rusefi/issues/6566#issuecomment-2166719232

commit d917d757 (2024-05-29): In Launch Control, add RPM Hysterisis fix #5610