rusefi/rusefi #5611: Improving current launch by configuring Smooth Retard

ElDominio, 2023-10-14:
**Is your feature request related to a problem? Please describe.**
I am frustrated when Smooth Retard only applies linearly from ???? to full retard.

**Would we need new configuration fields?**
Smooth Retard Start RPM (from launch) - At this value Ignition Timing **starts** being moved from the main table to the "Ignition Retard(deg) value in the Launch Control Menu. Assuming our launch RPM is 4500, and this value is 500, our Ignition Timing starts being biased toward Ignition Retard(deg) at 4000RPM.
Smooth Retard End RPM (from Launch) - At this value Ignition  Timing **finishes** being moved from the main table to the "Ignition Retard(deg) value in the Launch Control Menu. Assuming our launch RPM is 4500, and this value is 100, our Ignition Timing **ENDS** being biased toward Ignition Retard(deg) at 4400RPM.

**Describe the solution you'd like**
I would like to be able to manage when the retarding takes place and why.

kifir23917, 2024-05-24:
And another question is about the case when:
1. `Ignition Retard enable` is `true`
2. `Smooth Retard Mode` is `false`.

As I understand in this case I need simply to replace `getRunningAdvance` function result with `Ignition Retard(deg)` parameter value without any interpolation. Is my understanding correct, that I need to replace this result only when RPM sensor value is in the window specified by `Launch RPM(rpm)`, `Smooth Retard Start RPM (flom launch)` and `Smooth Retard End RPM (flom launch)` parameters?

kifir23917, 2024-05-24:
Just for a reference I'm attaching a screenshot with the current parameters of `Launch Control`:
[image]

ElDominio, 2024-05-24:
> Currently we have the function `getRunningAdvance` that calculates `ignition timing angle advance before TDC` from `rpm` and `engineLoad`. When:
> 
> 1. `Ignition Retard enable` is `true`
> 2. `Smooth Retard Mode` is `true`
> 3. launch condition is satisfied,
> 
> current code replaces `getRunningAdvance` function result with the value marked with pink circle on the following picture: ![rusefi-Current](https://private-user-images.githubusercontent.com/159958110/333690320-3e94f80e-b647-4a4a-b492-d8fedebfdff6.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY1ODI0MDMsIm5iZiI6MTcxNjU4MjEwMywicGF0aCI6Ii8xNTk5NTgxMTAvMzMzNjkwMzIwLTNlOTRmODBlLWI2NDctNGE0YS1iNDkyLWQ4ZmVkZWJmZGZmNi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUyNFQyMDIxNDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mZGRlYzhhNWI3NjkwZjdkZWNhMTJmMGY0NjdhY2U3ODM1NjUzODg2YWFhY2I2NDJjZDc0OGNkYjY2M2UwNzcyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.9p8a452w1nGccWKW8sAWWmRVk4ueLJgUh9_OCTp1oDY)
> 
> On the picture above:
> 
> * square brackets are the value of the corresponding `Launch Control` parameter
> * `{current rpm}` is value from RPM sensor
> * `{current ignition timing angle advance without retard}` is the value that function `getRunningAdvance` would return if both `Ignition Retard enable` and `Smooth Retard Mode` are `false`.
> 
> This code looks incorrect and for the case when:
> 
> 1. `Ignition Retard enable` is `true`
> 2. `Smooth Retard Mode` is `true`
> 3. RPM sensor value is in the window specified by `Launch RPM(rpm)`, `Smooth Retard Start RPM (flom launch)` and `Smooth Retard End RPM (flom launch)` parameters
> 
> I'm going to replace `getRunningAdvance` function result with the value marked with pink circle on the following picture: ![rusefi-Proposal](https://private-user-images.githubusercontent.com/159958110/333692307-2c3ada4a-4239-4cbe-84d0-b5bb2a5089db.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY1ODI0MDMsIm5iZiI6MTcxNjU4MjEwMywicGF0aCI6Ii8xNTk5NTgxMTAvMzMzNjkyMzA3LTJjM2FkYTRhLTQyMzktNGNiZS04NGQwLWI1YmIyYTUwODlkYi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUyNFQyMDIxNDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zY2I3YTgxMjAxZTAyYTVlNjdlZjM3MWJkZjJhN2RmNWJkMzRmZDkyOGVjMjEyMDZiZjkxYjcwNzY2NjY4ZWRkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.JSqyN1SJpmlIOeyflDj9m64My7hKciemzgz9vCSG7qQ)
> 
> @ElDominio, could you please confirm that it is what you want?

This sounds exactly like what I am looking for. The only thing is that I think Smooth Retard Start RPM and Launch Control Window will be the same number in every case, meaning one of them can be removed.

The graph shows pretty much what I meant, using your reference values in

> Just for a reference I'm attaching a screenshot with the current parameters of `Launch Control`: ![Launch Control](https://private-user-images.githubusercontent.com/159958110/333697066-01296f01-b472-4e69-aebf-6890d57467d0.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY1ODI0MDMsIm5iZiI6MTcxNjU4MjEwMywicGF0aCI6Ii8xNTk5NTgxMTAvMzMzNjk3MDY2LTAxMjk2ZjAxLWI0NzItNGU2OS1hZWJmLTY4OTBkNTc0NjdkMC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUyNFQyMDIxNDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wMDg5YzNlMWZmNGVjMjViYjBiY2IwZmFiMWEzNTlkMWQxYjU2YWQzN2YyNjhjMmZkZGQwZmYyNTI2YmE4ZjBhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ep-Pe1FC8pNbXFDR0i38nz7zH6sdWOhYu82M3wMJq5E)

I would understand that at 500rpm BEFORE launch, the ignition starts changing from getRunningAdvance value to the "Ignition Retard(deg)" value, and at 100rpm BEFORE launch, it has arrived at the "Ignition Retard(deg)" value shown to be 0. Then from that point until the "Launch RPM" Advance will be at the value in "Ignition Retard".

I only reiterate this to make sure I understood your graphs, because I think that's what it shows

> And another question is about the case when:
>
> 1. `Ignition Retard enable` is `true`
> 2. `Smooth Retard Mode` is `false`.
>
> As I understand in this case I need simply to replace `getRunningAdvance` [...]

Correct

kifir23917, 2024-05-24:
> The only thing is that I think Smooth Retard Start RPM and Launch Control Window will be the same number in every case, meaning one of them can be removed.
So we are going to reuse the same `Launch Control Window(RPM)` parameter in two different ways:
1. as `Launch RPM Hysteresis`
2. as `Smooth Retard Start RPM (from launch)`

and we do not expect that we can want to use different values for cases (1) and (2) above. Am I correct?

> at 100rpm BEFORE launch, it has arrived at the "Ignition Retard(deg)" value shown to be 0. Then from that point until the "Launch RPM" Advance will be at the value in "Ignition Retard".

So, as I understood, a behavior with `Smooth Retard Mode` = `false` will be absolutely the same as a behavior with:
1. `Smooth Retard Mode` = `true`
2. `Launch Control Window(RPM)` = `Smooth Retard End RPM (from launch)`

Final algorithm of `getRunningAdvance` function will be the following:
[image]

kifir23917, 2024-05-25:
@ElDominio, I suspect that I found another problem.
Current code calculates `{current ignition timing angle advance without retard}` using `{current rpm}` value. For me it looks strange that we use value calculated from `{current rpm}` as interpolation value for `[Launch PRM(rpm)] - [Launch Control Window(RPM)]`. I'm going to use `[Launch PRM(rpm)] - [Launch Control Window(RPM)]` instead of `{current rpm}` to calculate `{current ignition timing angle advance without retard}` in case when:
- `Ignition Retard enable` is `true`
- `Smooth Retard Mode` is `true`
- RPM sensor value is in the window specified by `Launch RPM(rpm)`, `Launch Control Window(RPM)` and `Smooth Retard End RPM (flom launch)` parameters.

Please notify me if I'm missing anything and you have any objections.

ElDominio, 2024-05-26:
I think it should be current RPM, since using currentRPM make the interpolation smooth from the actual timing table to the target timing retard instead of being a flat line (with no beginning? i think i need to test this in a real vehicle to understand what you mean)

kifir23917, 2024-05-27:
Well, I'm leaving using {current rpm} for smooth interpolation.

ElDominio, 2024-05-27:
Current logic seems to be from testing today:
Spark is cut when currentRPM > (Launch RPM + Launch Control Window) meaning that in this car where Launch RPM is 2500 and Window is 500, spark is cutting at 3000 when it should cut at 2500.
Spark should be cut when currentRPM > launchRPM.

The Window should be the RPM _before_ the Launch RPM where corrections start taking place.

__________________________________________________________
Timing does not change now, no matter the value in "Ignition Retard"  and whether or not "Ignition Retard Enable"  is true or not.

[LaMitsubichaReloaded.zip](https://github.com/rusefi/rusefi/files/15461833/LaMitsubichaReloaded.zip)

log and tune, showing settings and incorrect cut behavior

kifir23917, 2024-05-27:
This problem is fixed in scope of #6533 - the fix has been just merged into master.

ElDominio, 2024-05-28:
Behavior is now much better, where the correction starts applying at Launch Control Window, and maximum correction is applied at "Smooth retard end rpm"

The only problem now is that, we have to stay at the "Ignition Retard" value in any value OVER the Smooth Retard End RPM

For example
[image]

[timing jumps back.zip](https://github.com/rusefi/rusefi/files/15474781/timing.jumps.back.zip)

kifir23917, 2024-05-29:
Requirements from Telegram:
> the only thing left to improve in the current implementation is that timing should stay at the "ignition retard" after the end RPM for smooth timing retard

> the idea is to transition slowly from the window into the max retard, then keep increasing RPM until we reach cut, where we cut

kifir23917, 2024-05-29:
> Should timing stay at the Ignition Retard(deg) level after rpm overcomes Launch RPM(rpm) setting? Or do I need just to fix the code to stay on  Ignition Retard(deg) level when RPM is between  Smooth Retard End RPM (from launch) and  Launch RPM(rpm) settings?

> it should yes, since we are still inside the "launch" state.
> timing should stay at the Ignition Retard(deg) level after rpm overcomes Launch RPM(rpm) setting

kifir23917, 2024-05-29:
kifir23917:
> I have a question about case when Ignition Retard enable is true and Smooth Retard Mode is false. > Starting what rpm timing should stay at the Ignition Retard(deg) level? I see the following options:
> 1.  [`Launch RPM(rpm)`]
> 2.  [`Launch RPM(rpm)`] - [`Launch Control Window(RPM)`]
> 3.  [`Launch RPM(rpm)`] - [`Smooth Retard End RPM (from launch)`]
>
> And the first one looks for me like the most natural. Am I correct?

ElDominio:
> Nope, should be option 2

PR #6562 (2024-05-29): add references to requirements in test
closes #5611
commit afe07218 (2024-05-29): add references to requirements in test
closes #5611
closed 2024-05-29 by rusefillc