rusefi/rusefi #5612: Brain dump: Better Launch Control cut

ElDominio, 2023-10-14:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when Launch Control can only fully cut ignition at the launch control RPM value, then revive it at some distance from the launch RPM. We should be able to try to make RPM hold as steady as possible close to this.

**Would we need new configuration fields?**
Variable Cut Enable - This allows the user to select if they want to vary the amount  of ignition events near the cut, or just use a simple on/off limiter.
Variable Cut Type - If we want a variable cut, are we implementing it with a PID or manually?
________________________________________________________________ 

1. Manual Variable Cut - This mode allows the user to have something similar to the Smooth Retard mode, but cutting ignition events instead of retarding timing.
--Manual Variable Cut Start RPM - the RPM **from** the main Launch RPM where we begin to cut ignition events. If this value is 500 and Launch RPM is 4500, we start cutting Manual Variable Cut Start % of events from the ignition schedule at 4000RPM.
--Manual Variable Cut Start % - % of events to drop once we hit the remainder of (Launch RPM - Manual Variable Cut Start RPM). If this is 10%, cut 10% of ignition events.
--Manual Variable Cut End RPM - the RPM **from** the main Launch RPM where we begin to cut ignition events. If this value is 100 and Launch RPM is 4500, we start cutting Manual Variable Cut End % of events from the ignition schedule at 4400RPM. Should never be over 90%.
--Manual Variable Cut Start % - % of events to drop once we hit the remainder of (Launch RPM - Manual Variable Cut End RPM). If this is 90%, cut 90% of ignition events.
_________________________________________________________________________________

2. PID Cut -  This mode uses a PID to decide when to cut/not cut ignition. We feed the PID the current RPM, the Launch RPM as the target, and the % cut as the control variable (since changing % cut allows us to control the current RPM).
--PID Start RPM - the RPM **from** the Launch RPM when the PID can start applying cuts to ignition to control the current RPM. We have this because if not the PID might just start cutting events when RPM is rising to the cut, but way too early.
--P - proportional cut % in the error in RPM 
--I - how much % we cut/not cut additionally because we are far/close from the RPM target
--D - how much  % we cut/not cut because we are rising too fast/not rising at all
________________________________________________________________

**Describe the solution you'd like**
At least a discussion on how we can get a slightly smarter launch control to work

ElDominio, 2023-10-30:
https://youtu.be/bIxulvBHTFo

This video explains some stuff about building boost, and traction control. Good watch for the topic on this thread till about 5:30

rusefillc, 2024-05-14:
See also https://www.rusefi.com/forum/viewtopic.php?f=5&t=2670

ElDominio, 2024-06-14:
this ticket kinda sucks looking back on it but first part was done, and PID cut is a bit more complex than what i explained

closed 2024-06-14 by ElDominio