rusefi/rusefi #5616: Remove Injector Test from "Controller"

ElDominio, 2023-10-16:
**Describe the bug**
Why do we have "Injector Test" when we can test injector in Bench Test with the same parameters?

**Hardware**
 - Board: [e.g. microRusEFI]
 - Engine info [e.g. m111 NA]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Controller'
2. Click on 'Injector Test'
4. See error

**Expected behavior**
Only have one option, either in Bench Test or Injector Test (I like bench test better)

**Screenshots**
[image]

mck1117, 2023-10-16:
Did you actually look at each? The injector test page has stuff to facilitate both testing injectors and characterizing injectors.

ElDominio, 2023-10-16:
Yes, and I didn't understand what additional data from what I have on the main dashboard/panel and how is it useful (I can look at battery in the gauges and what are test iterations)

If it's that important for such an edge case, it should be in a folder that whoever uses that specific function can find it; most end users end up asking "which test mode do I use?" even if they're are to test specifically coils.

ElDominio, 2024-04-06:
Apparently not a good idea so closing

closed 2024-04-06 by ElDominio