rusefi/rusefi #5622: Password protection makes ECU go CRITICAL

ElDominio, 2023-10-19:
**Describe the bug**
Adding password protection makes the EXU pretty much untuneable, since it always throws a "password protection is activated" critical

**Hardware**
 -firmware

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Experimental'
2. Click on 'Tune protection'
3. Add password
4. Disconnect and Reboot
5. Connect
6. Try to change a value
7. Critical error

**Expected behavior**
I can modify tune without problem

**Additional crap**
Can this be done from console? Have console be the "door lock" to lock/unlock the tune? Example, when tuner wants to lock the tune, they do it from Console, then it becomes unresponsive to TS, and vice versa. (I only suggest this if it makes implementation easier)

rusefillc, 2023-10-19:
Expected behavior, poor usability.

[image]

ElDominio, 2023-10-19:
I guess I failed at problem statement, i had no idea how to unlock. Was it in console all along?

rusefillc, 2023-10-19:
I do not see documentation on https://github.com/rusefi/rusefi/issues/4243

Your confusion is well justified - it looks like the unlocking mechanism was never documented :(

rusefillc, 2023-10-19:
@ElDominio while I am keeping this ticket open while working on documentation could it also be that you were looking for two SEPARATE levels of protection - one for READ/WRITE and another one only for WRITE protection? I.e. what was you assumption to have tuneability while having password?

Maybe you were assuming that since your TS knows password it would be tuneable. Nope, that's not how it works :( You lock from everyone.

One of confusions come from the fact that project you had used to lock still has tune on PC side. So on that PC it looks like you have access while all you have is local copy from before locking.

rusefillc, 2023-10-20:
@ElDominio https://github.com/rusefi/rusefi/wiki/Password-Protection

closed 2023-10-20 by rusefillc