rusefi/rusefi #5634: AEM CAN wideband not working since October

ElDominio, 2023-10-24:
**Describe the bug**
A car had working CAN Wideband with an AEM X Series, updated firmware and it no longer works

**Hardware**
4chan RevH

**To Reproduce**
Steps to reproduce the behavior:
Have working CAN wideband
update firmware
no longer have working wideband

**Expected behavior**
Wideband reading correctly when CAN is correctly hooked up

rusefillc, 2023-11-01:
@ElDominio @Ankyth need help isolation the issue

step 1: can we find ANY release from the last year when this worked? https://github.com/rusefi/rusefi/releases We have to confirm we have ANY date when this has worked

step 2-5: once we have release which is working please find middle of "working and not working" dates and try isolating problem to a specific day?

Ankyth, 2023-11-02:
@rusefillc  my tune has date of: (rusEFI v20230803@64196) and the canbus works

ElDominio, 2023-11-02:
@Ankyth means his firmware is v20230803

@Ankyth can you update and see if it breaks?

rusefillc, 2023-12-04:
Advanced troubleshooting:
rusEFI console
``enable verbosecan``

```
2023-12-03_20_01_06_141: EngineState: CAN0 RX: ID 0000180 DLC 8: 3F 2A 1D A9 82 4A 91 06
2023-12-03_20_01_06_143: EngineState: CAN0 RX: ID 0000180 DLC 8: 3F 84 1D F6 82 4B 91 06
2023-12-03_20_01_06_145: EngineState: CAN0 RX: ID 0000180 DLC 8: 3F F2 1E 53 82 4B 91 06
2023-12-03_20_01_06_147: EngineState: CAN0 RX: ID 0000180 DLC 8: 3F 70 1D E5 82 4A 91 06
2023-12-03_20_01_06_151: EngineState: CAN0 RX: ID 0000180 DLC 8: 3F DE 1E 42 82 4B 91 06
2023-12-03_20_01_06_151: EngineState: BENCH_TEST Sending CAN bus0 message: ID=770000/l=8 0 0 0 0 0 0 0 0
2023-12-03_20_01_06_151: EngineState: BENCH_TEST Sending CAN bus0 message: ID=770001/l=8 0 0 0 0 0 E9 E9 0
2023-12-03_20_01_06_151: EngineState: BENCH_TEST Sending CAN bus0 message: ID=770006/l=8 0 0 E9 E9 0 0 0 0
2023-12-03_20_01_06_151: EngineState: WBO_SERVICE Sending CAN bus0 message: ID=EF50000/l=2 1 0 0 0 0 0 0 0
2023-12-03_20_01_06_151: EngineState: CAN0 RX: ID 0000180 DLC 8: 3F DE 1E 42 82 4B 91 06
2023-12-03_20_01_06_151: EngineState: CAN0 RX: ID 0000180 DLC 8: 40 56 1E A9 82 4A 91 06
2023-12-03_20_01_06_151: EngineState: CAN0 RX: ID 0000180 DLC 8: 40 2E 1E 86 82 4A 91 06
2023-12-03_20_01_06_151: EngineState: CAN0 RX: ID 0000180 DLC 8: 40 24 1E 7E 82 4A 91 06
2023-12-03_20_01_06_151: EngineState: CAN0 RX: ID 0000180 DLC 8: 40 6A 1E BA 82 4A 91 06
```

ElDominio, 2023-12-04:
[image]

[image]

[image]

on 20231204 firmware
hellen154hyundai

[noCANaemwideband.zip](https://github.com/rusefi/rusefi/files/13551883/noCANaemwideband.zip)

[CurrentTune.zip](https://github.com/rusefi/rusefi/files/13551897/CurrentTune.zip)

rusefillc, 2023-12-04:
@ElDominio  please go  advanced troubleshooting:
rusEFI console
``enable verbosecan``

ElDominio, 2024-04-06:
More magic. Tested a wideband yesterday and it works

closed 2024-04-06 by ElDominio