rusefi/rusefi #5641: Non-square dimension maps

pchmura4, 2023-10-25:
**Describe the bug**
When the X and Y axis of 3D map are not equal, the interpolation results are not correct.

**Hardware**
All?

**To Reproduce**
Create a non-square 3D map that will be generated from rusefi_config.txt

**Expected behavior**
Correct interpolation results

**Screenshots**
N/A

**Additional context**
Probably the issue could be in the script that generates board *.ini files for TunerStudio.
Swapping dimensions of 3D table seems to be solving the issue, needs additional checks.

rusefillc, 2023-10-25:
@pchmura4 let's reproduce with a test

For something like this unit test should be suitable?

rusefillc, 2023-10-25:
Either existing interpolation test or we can hard code something into actual rusEFI structures

pchmura4, 2023-10-25:
I think that unit test will pass, as the interpolation function seems to be good, for me it looks like the values from TunerStudio are sent in incorrect order.
The screenshot below is probably numerically wrong but maybe it can give a little hint.
[image]
For me it looks like the map in memory after being received from TS is malformed after being "transposed".
I can be wrong, and imagining things, did not have too much of time to give it proper testing

rusefillc, 2023-10-25:
@pchmura4 please hard-code any test data into default_base_engine.cpp any non-square table under ``#if EFI_SIMULATOR`` conditional compilation? this was we would see real TS connect to official simulator binary and load whatever we have hard-coded?

just pushed minor naming improvement for the test, order of arguments if definitely weird but that's a separate minor issue

rusefillc, 2023-10-25:
hmm :(

https://github.com/rusefi/rusefi/commit/77b7cd683d54ffb3c2c6b63d0e341b944b4028b6

[image]

rusefillc, 2023-10-26:
https://github.com/rusefi/rusefi/commit/9d0eac561e794bbb3ba915a42a8640c64022df86

[image]

rusefillc, 2023-10-26:
my fix flips TS definition dimensions

looks better with the fix?

[image]

mck1117, 2023-10-26:
the question is not whether you can set a default properly

the question is whether the LOOKUP works properly in the ECU to agree with the axes on TS

a table in TS that looks like 

1 2 3
4 5 6
7 8 9

is stored in memory as 7 8 9 4 5 6 1 2 3 (start bottom left corner, move right then up)

pchmura4, 2023-10-26:
how can I build for the simulator so I can make adjustments and test it myself?

mck1117, 2023-10-26:
> how can I build for the simulator so I can make adjustments and test it myself?

```
cd ./simulator
make
```

rusefillc, 2023-10-26:
@pchmura4 please update source code, sometime master has typos looks like what you have

Also simulator windows binary is part of bundle

pchmura4, 2023-10-26:
Synced with latest changes, synced my fork, did "git reset --hard" and "git pull --rebase", still the same. Not possible to build on latest Debian Testing

rusefillc, 2023-10-26:
* https://github.com/rusefi/rusefi/ has a green checkbox
* https://github.com/rusefi/rusefi/blob/master/.github/workflows/build-simulator.yaml has the steps GHA takes
* https://github.com/pchmura4/rusefi/ is in sync but no GHA open question if you want to enable GHA on your fork
* could this be a case of an old/wrong in some way gcc or something?

pchmura4, 2023-10-26:
alright, I'll take a look at the yaml and try to reproduce it myself

rusefillc, 2023-10-26:
@pchmura4 let's move that to https://github.com/rusefi/rusefi/issues/5644

pchmura4, 2023-10-27:
Hi, I have gave it a try and it looks to be working correctly :)
[image]
[image]

closed 2023-10-27 by rusefillc