rusefi/rusefi #5644: simulator does not compile for pchmura4

rusefillc, 2023-10-26:
Do we need something similar to [gcc_version_check.c](https://github.com/rusefi/rusefi/blob/master/firmware/gcc_version_check.c) in simulator?

rusefillc, 2023-10-26:
@pchmura4 can you please ``gcc -v``?

here is my happy debian-based mxlinux

```
Andrey@nuc1:~/rusefi/simulator
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/10/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa:hsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 10.2.1-6' --with-bugurl=file:///usr/share/doc/gcc-10/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-10 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-10-Km9U7s/gcc-10-10.2.1/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-10-Km9U7s/gcc-10-10.2.1/debian/tmp-gcn/usr,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-mutex
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.2.1 20210110 (Debian 10.2.1-6)
Andrey@nuc1:~/rusefi/simulator
```

pchmura4, 2023-10-26:
`❯ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-linux-gnu/13/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 13.2.0-5' --with-bugurl=file:///usr/share/doc/gcc-13/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-13 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/libexec --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --enable-cet --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/reproducible-path/gcc-13-13.2.0/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/reproducible-path/gcc-13-13.2.0/debian/tmp-gcn/usr --enable-offload-defaulted --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-serialization=2
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 13.2.0 (Debian 13.2.0-5)`

pchmura4, 2023-10-26:
Looks like newer gcc is more strict with non constexpr variables inside constexpr.
Changing these two made the compilation succeed now

[image]
[image]

Got it to work by changing these two lines now 😊

rusefillc, 2023-10-26:
nice! let me see if I would be able to reproduce

rusefillc, 2023-10-26:
https://github.com/rusefi/rusefi/pull/5645 currently does not compile because I am a weak human who fails to OCR a diff in the screen shot form. 

Branches and commits in text form are definitely a more effective way to share diffs.

pchmura4, 2023-10-26:
sorry, git is such a pain for me, so I just shared a screenshot, so you don't get 20 commits from previous work at the same time

`diff --git a/firmware/console/binary_log/binary_logging.cpp b/firmware/console/binary_log/binary_logging.cpp
index 557bc3599c..78e2c06310 100644
--- a/firmware/console/binary_log/binary_logging.cpp
+++ b/firmware/console/binary_log/binary_logging.cpp
@@ -47,7 +47,7 @@ void writeSdLogLine(Writer& bufferedWriter) {
 	binaryLogCount++;
 }
 
-static constexpr uint16_t recordLength = computeFieldsRecordLength();
+static const uint16_t recordLength = computeFieldsRecordLength();
 
 void writeFileHeader(Writer& outBuffer) {
 	char buffer[MLQ_HEADER_SIZE];
diff --git a/firmware/console/binary_log/log_fields_generated.h b/firmware/console/binary_log/log_fields_generated.h
index fb5c957068..7e61233b57 100644
--- a/firmware/console/binary_log/log_fields_generated.h
+++ b/firmware/console/binary_log/log_fields_generated.h
@@ -1,4 +1,4 @@
-static constexpr LogField fields[] = {
+static const LogField fields[] = {
 {packedTime, GAUGE_NAME_TIME, "sec", 0},
 	{engine->outputChannels.RPMValue, "RPM", "RPM", 0},
 	{engine->outputChannels.rpmAcceleration, "dRPM", "RPM acceleration", 2},`

rusefillc, 2023-10-26:
git being a pain in a sign of not using some great git client with UI

i believe that JetBrains git client shipped as part of [IDEA](https://www.jetbrains.com/idea) or Android Studio is the best git client ever. See https://youtrack.jetbrains.com/issue/IDEA-152437/Make-git-client-a-standalone-app :)

pchmura4, 2023-10-26:
now that makes life easier :)

rusefillc, 2024-01-02:
Something similar in the firmware https://rusefi.com/forum/viewtopic.php?p=48876#p48876