rusefi/rusefi #5655: Volvo 240 60-2 trigger with holes instead of teeth

ElDominio, 2023-10-27:
**Describe the bug**
Tooth time looks as if it was a 60-1, but it looks like a 60-2 and we counted 58 teeth (holes), similar to picture but instead of having a 2x air gap it is 2x solid.

Log of gap is at: https://rusefi.com/online/view.php?log=1313

**Hardware**
-paralela
-max9924 vr to hall
-volvo 240 turbo '89

**To Reproduce**
start engine
rpm reading not consistent

**Expected behavior**
engine fires? the wiring is garbage but i'm just making a record

**Screenshots**
[image]

**Additional context**
this is just to log it, but not critical, the wiring is extremely poor in this car

mck1117, 2023-10-27:
I ran a volvo 240 for a long time with the stock trigger setup with no issue, on a proteus with max9924. Swap your VR wires.

ElDominio, 2023-10-28:
I didn't try that one lol, thanks @mck1117 i'll try it monday

the wiring on it is horrifying so i don't rule out noise either, and the flywheel is warped (you can physically see it wobble up and down while cranking, and it reflects on the oscope) so i told him to get a new one

ElDominio, 2023-10-30:
@mck1117 it worked, thanks!

also had to use custom toothed wheel instead of 60/2 but w/e it ran jsut fine even with shit wiring

closed 2023-10-30 by ElDominio