rusefi/rusefi #5657: 36/2/2/2 EJ20 trigger issues

ElDominio, 2023-10-28:
**Describe the bug**
Trigger not inverted reads RPM, but horribly, and can never get to running
Log at: 
[notinverted36minus222.zip](https://github.com/rusefi/rusefi/files/13196880/notinverted36minus222.zip)

Trigger inverted gapRatio looks MUCH cleaner, but no RPM reading
log at: 
[invertedtrigger36minus222.zip](https://github.com/rusefi/rusefi/files/13196885/invertedtrigger36minus222.zip)

Custom trigger gaps get shot down by rusefi (why can't i write dumb numbers? is this not an advanced feature? why am i being babied)

**Hardware**
 - 4chan revE latest fw
 - EJ20

**Expected behavior**
Car runs properly

**Additional context**
Tune at: 
[36minus222.zip](https://github.com/rusefi/rusefi/files/13196890/36minus222.zip)

rusefillc, 2023-10-29:
@ElDominio I see 100 per second data rate and I say you should do better!

rusefillc, 2023-10-29:
@ElDominio have you used https://github.com/rusefi/rusefi/wiki/Trigger-Configuration-Guide#determine-correct-polarity to confirm polarity?

not inverted fails due to gaps being not useful

0.45 should be 0.38 and 2.7 should be 3.2

[image]

ElDominio, 2023-10-29:
> @ElDominio I see 100 per second data rate and I say you should do better!

Vendors usually have free edition...

rusefillc, 2023-10-29:
> Vendors usually have free edition...

please spank those vendors or get them a bulk license deal? but we can survive

rusefillc, 2023-10-29:
> Trigger inverted gapRatio looks MUCH cleaner, but no RPM reading 

I cannot explain that :( next step is rusEFI console log with detailed trigger sync checkbox, it would show more detailed info on why gaps were accepted or not.

rusefillc, 2023-10-29:
> Custom trigger gaps get shot down by rusefi (why can't i write dumb numbers? is this not an advanced feature? why am i being babied)

you are being vague. Do you mean critical error? what's the text of critical error? let's improve the text.

dumb numbers would sync more than once per revolution or never sync, those are rejected since would not function for sure

ElDominio, 2023-10-29:
> I cannot explain that :( next step is rusEFI console log with detailed trigger sync checkbox, it would show more detailed info on why gaps were accepted or not.

I'm first going to swap to MAX9924 because i hate discrete for high tooth counts

> you are being vague. Do you mean critical error? what's the text of critical error? let's improve the text.

dumb numbers would sync more than once per revolution or never sync, those are rejected since would not function for sure

I like that explanation a lot, even though I'm dumb it tells me exactly how i'm being dumb lol

ElDominio, 2024-02-23:
It was discrete issue because it is running on older hardware, added external 9924 and problem left