rusefi/rusefi #5661: HW CI instability: handleShaftSignal unexpected loop

rusefillc, 2023-10-29:
handleShaftSignal unexpected loop

```
 231029 165633.791 [main] IoUtil - Sending command [set engine_type 3]
I 231029 165633.792 [ECU Commands Queue] BinaryProtocol - Sending [set engine_type 3]
I 231029 165636.120 [ECU Commands Queue] IoUtil - Got confirmation in 2328ms
I 231029 165636.120 [main] IoUtil - Command [set engine_type 3] executed in 2329
2023-10-29_16_56_36_121<EOT>: Sleeping 1 seconds
I 231029 165636.127 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: applyNonPersistentConfiguration()
I 231029 165636.128 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: initializeTriggerWaveform(TT_FORD_ASPIRE/1)
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: Writing pending configuration...
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: FLASH_SUCCESS
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: NONE
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: NONE
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: NONE
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: NONE
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: NONE
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: NONE
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: NONE
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: NONE
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: PC4
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: PA3
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: PA6
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: PA5
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: NONE
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: NONE
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: NONE
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: NONE
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: NONE
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: NONE
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: NONE
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: NONE
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: NONE
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: NONE
I 231029 165636.129 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: unregistering PC15
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: PC15
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: unregistering PE6
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUnused: PE6
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUsed: Tps1Primary on PA3
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUsed: BatteryVoltage on PA0
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUsed: Clt on PC1
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUsed: AFR on PC4
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: extiTriggerTurnOnInputPin Trigger #1 PC6
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUsed: Trigger #1 on PC6
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: extiTriggerTurnOnInputPin Trigger #2 PA5
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUsed: Trigger #2 on PA5
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: pin_markUsed: trigger debug on PC15
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: AUX Serial started
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: initializeTriggerWaveform(TT_FORD_ASPIRE/1)
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: Template FORD_ASPIRE_1996/3 trigger TT_FORD_ASPIRE/LM_SPEED_DENSITY/0
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: configurationVersion=18
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: rpmHardLimit: 7000/operationMode=2
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: globalTriggerAngleOffset=175.00
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: === cranking ===
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: crankingRpm: 550
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: cranking injection IM_SIMULTANEOUS
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: cranking timing -37.00
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: === ignition ===
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: ignitionMode: IM_ONE_COIL/enabled=Yes
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: timingMode: TM_DYNAMIC
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: === injection ===
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: injection IM_SEQUENTIAL enabled=Yes
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: injectionPins: mode OM_DEFAULT
I 231029 165636.130 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: injection #1 @ NONE
I 231029 165636.131 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: injection #2 @ NONE
I 231029 165636.131 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: injection #3 @ NONE
I 231029 165636.131 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: injection #4 @ NONE
I 231029 165636.131 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: ignitionPins: mode OM_DEFAULT
I 231029 165636.131 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: ignition #1 @ NONE
I 231029 165636.131 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: ignition #2 @ NONE
[... 33 lines cut ...]
I 231029 165637.219 [main] IoUtil - Sending command [set_sensor_mock 4 69.12]
I 231029 165637.219 [ECU Commands Queue] BinaryProtocol - Sending [set_sensor_mock 4 69.12]
I 231029 165637.220 [ECU Commands Queue] IoUtil - Got confirmation in 1ms
I 231029 165637.220 [main] IoUtil - Command [set_sensor_mock 4 69.12] executed in 1
I 231029 165637.221 [main] IoUtil - Sending command [set_sensor_mock 30 12]
I 231029 165637.221 [ECU Commands Queue] BinaryProtocol - Sending [set_sensor_mock 30 12]
I 231029 165637.222 [ECU Commands Queue] IoUtil - Got confirmation in 1ms
I 231029 165637.222 [main] IoUtil - Command [set_sensor_mock 30 12] executed in 1
[... cut ...]
I 231029 165713.328 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: confirmation_set_individual_coils_ignition:29
I 231029 165713.328 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: confirmation_reset_engine_chart:18
I 231029 165713.328 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: setTpsAccelLen: Length should be positive [0]
I 231029 165713.328 [ECU Communication Executor1] MessagesCentral - postMessage EngineState: CRITICAL error: handleShaftSignal unexpected loop end 20231029@stm32f407vg
```

rusefillc, 2024-04-21:
No, still an issue

```
	at org.junit.runner.JUnitCore.run(JUnitCore.java:94)
	at com.rusefi.CmdJUnitRunner.runHardwareTest(CmdJUnitRunner.java:26)
	at com.rusefi.CmdJUnitRunner.runHardwareTestAndExit(CmdJUnitRunner.java:10)
	at com.rusefi.HwCiF4Discovery.main(HwCiF4Discovery.java:40)
test(com.rusefi.f4discovery.VssHardwareLoopTest): Not true: Fatal not expected: CRITICAL error: handleShaftSignal unexpected loop end 20240421@community
Test failed: test(com.rusefi.f4discovery.VssHardwareLoopTest) because Not true: Fatal not expected: CRITICAL error: handleShaftSignal unexpected loop end 20240421@community
Exception:
testV12(com.rusefi.f4discovery.HighRevTest): Not true: Fatal not expected: CRITICAL error: handleShaftSignal unexpected loop end 20240421@community
Test failed: testV12(com.rusefi.f4discovery.HighRevTest) because Not true: Fatal not expected: CRITICAL error: handleShaftSignal unexpected loop end 20240421@community
Exception:
testVW(com.rusefi.f4discovery.HighRevTest): Not true: Fatal not expected: CRITICAL error: handleShaftSignal unexpected loop end 20240421@community
Test failed: testVW(com.rusefi.f4discovery.HighRevTest) because Not true: Fatal not expected: CRITICAL error: handleShaftSignal unexpected loop end 20240421@community
```

rusefillc, 2024-05-03:
problem: we do not have S2R

@kifir23917 let's add some assertions and wait for next occurrence? :(

rusefillc, 2024-05-07:
HwCiF4Discovery.java

rusefillc, 2024-05-12:
```
java.lang.AssertionError: Not true: Fatal not expected: CRITICAL error: handleShaftSignal unexpected loop end 2 10 545.000000000 20240511@community
	at org.junit.Assert.fail(Assert.java:89)
	at com.rusefi.TestingUtils.assertTrue(TestingUtils.java:34)
	at com.rusefi.TestingUtils.assertNull(TestingUtils.java:104)
	at com.rusefi.functional_tests.TestHelper.assertNotFatal(TestHelper.java:21)
	at com.rusefi.functional_tests.EcuTestHelper.sendCommand(EcuTestHelper.java:99)
	at com.rusefi.functional_tests.EcuTestHelper.sendCommand(EcuTestHelper.java:95)
	at com.rusefi.functional_tests.EcuTestHelper.createInstance(EcuTestHelper.java:86)
	at com.rusefi.RusefiTestBase.startUp(RusefiTestBase.java:17)
	at jdk.internal.reflect.GeneratedMethodAccessor1.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:59)
	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:56)
	at org.junit.internal.runners.statements.RunBefores.invokeMethod(RunBefores.java:33)
```

kifir23917, 2024-05-15:
By the moment we meet this issue only `During start-up`:
- https://github.com/rusefi/rusefi/actions/runs/9099107159/job/25010894605?pr=6518

commit 27081149 (2023-10-29): HW CI instability: handleShaftSignal unexpected loop fix #5661

shoot from the hip?