rusefi/rusefi #5701: Something kia rio starter 4chan

rusefillc, 2023-11-08:


ElDominio, 2023-11-08:
On 4chan + gdi, controlling starter does not work correctly

[image]

The start/stop button input triggers every X amount of seconds, exactly the same as set in that dialog box.
This happens regardless of reality with button input, which makes the engine shut off

rusefillc, 2023-11-09:
https://github.com/rusefi/rusefi/commit/8e458f4ec6f3fe3a08fc4f742d6918993a194183

rusefillc, 2023-11-09:
https://github.com/rusefi/rusefi/commit/50f9fce660a641b745d3f4835b644c98914429e9

rusefillc, 2023-11-09:
```
2023-11-08_22_40_34_582: EngineState: ac_switch timeLast 0
2023-11-08_22_40_34_587: EngineState: physical state 0 value 0
2023-11-08_22_40_34_587: EngineState: start_button timeLast 134095036
2023-11-08_22_40_34_588: EngineState: physical state 0 value 0
2023-11-08_22_40_34_590: EngineState: confirmation_debounce:8
2023-11-08_22_40_34_682: EngineState: Cranking timeout 1 seconds
2023-11-08_22_40_34_782: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
2023-11-08_22_40_35_585: EngineState: ac_switch timeLast 0
2023-11-08_22_40_35_586: EngineState: physical state 0 value 0
2023-11-08_22_40_35_587: EngineState: start_button timeLast 138495037
2023-11-08_22_40_35_587: EngineState: physical state 0 value 0
2023-11-08_22_40_35_588: EngineState: confirmation_debounce:8
2023-11-08_22_40_35_791: EngineState: Cranking timeout 1 seconds
2023-11-08_22_40_35_888: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
2023-11-08_22_40_36_692: EngineState: ac_switch timeLast 0
2023-11-08_22_40_36_693: EngineState: physical state 0 value 0
2023-11-08_22_40_36_693: EngineState: start_button timeLast 142895042
2023-11-08_22_40_36_696: EngineState: physical state 0 value 0
2023-11-08_22_40_36_705: EngineState: confirmation_debounce:8
2023-11-08_22_40_36_900: EngineState: Cranking timeout 1 seconds
2023-11-08_22_40_36_901: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
2023-11-08_22_40_37_956: EngineState: Cranking timeout 1 seconds
2023-11-08_22_40_37_995: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!

```

rusefillc, 2023-11-09:
[image]

rusefillc, 2023-11-09:
```
2023-11-08_22_51_18_648: EngineState: ac_switch timeLast 0
2023-11-08_22_51_18_649: EngineState: physical state 0 value 0
2023-11-08_22_51_18_649: EngineState: mode 0
2023-11-08_22_51_18_650: EngineState: start_button timeLast 0
2023-11-08_22_51_18_652: EngineState: physical state 0 value 0
2023-11-08_22_51_18_657: EngineState: mode 0
2023-11-08_22_51_18_659: EngineState: confirmation_debounce:8
2023-11-08_22_51_21_148: EngineState: ac_switch timeLast 0
2023-11-08_22_51_21_148: EngineState: physical state 0 value 0
2023-11-08_22_51_21_149: EngineState: mode 0
2023-11-08_22_51_21_149: EngineState: start_button timeLast 0
2023-11-08_22_51_21_149: EngineState: physical state 0 value 0
2023-11-08_22_51_21_152: EngineState: mode 0
2023-11-08_22_51_21_157: EngineState: confirmation_debounce:8
2023-11-08_22_51_25_692: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
2023-11-08_22_51_26_148: EngineState: Firing priming pulse of 18.64 ms
2023-11-08_22_51_26_669: EngineState: Cranking timeout 1 seconds
2023-11-08_22_51_26_748: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
2023-11-08_22_51_27_788: EngineState: Cranking timeout 1 seconds
2023-11-08_22_51_27_789: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
2023-11-08_22_51_28_888: EngineState: Cranking timeout 1 seconds
2023-11-08_22_51_28_890: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
2023-11-08_22_51_29_992: EngineState: Cranking timeout 1 seconds
2023-11-08_22_51_29_993: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
2023-11-08_22_51_31_095: EngineState: Cranking timeout 1 seconds
2023-11-08_22_51_33_648: EngineState: ac_switch timeLast 0
2023-11-08_22_51_33_649: EngineState: physical state 0 value 0
2023-11-08_22_51_33_650: EngineState: mode 0
2023-11-08_22_51_33_655: EngineState: start_button timeLast 70895022
2023-11-08_22_51_33_656: EngineState: physical state 0 value 0
2023-11-08_22_51_33_658: EngineState: mode 0
2023-11-08_22_51_33_659: EngineState: confirmation_debounce:8
2023-11-08_22_51_36_291: EngineState: Skipped priming pulse since ignSwitchCounter = 1
2023-11-08_22_51_36_292: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
2023-11-08_22_51_37_283: EngineState: Cranking timeout 1 seconds
2023-11-08_22_51_37_353: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
2023-11-08_22_51_37_951: EngineState: ac_switch timeLast 0
2023-11-08_22_51_37_952: EngineState: physical state 0 value 0
2023-11-08_22_51_37_953: EngineState: mode 0
2023-11-08_22_51_37_953: EngineState: start_button timeLast 100295032
2023-11-08_22_51_37_954: EngineState: physical state 0 value 0
2023-11-08_22_51_37_954: EngineState: mode 0
2023-11-08_22_51_37_957: EngineState: confirmation_debounce:8
2023-11-08_22_51_38_376: EngineState: Cranking timeout 1 seconds
2023-11-08_22_51_38_450: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
2023-11-08_22_51_39_350: EngineState: ac_switch timeLast 0
2023-11-08_22_51_39_351: EngineState: physical state 0 value 0
2023-11-08_22_51_39_351: EngineState: mode 0
2023-11-08_22_51_39_351: EngineState: start_button timeLast 104695026
2023-11-08_22_51_39_352: EngineState: physical state 0 value 0
2023-11-08_22_51_39_352: EngineState: mode 0
2023-11-08_22_51_39_353: EngineState: confirmation_debounce:8
2023-11-08_22_51_39_453: EngineState: Cranking timeout 1 seconds
2023-11-08_22_51_39_552: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
2023-11-08_22_51_40_584: EngineState: Cranking timeout 1 seconds
2023-11-08_22_51_40_652: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
2023-11-08_22_51_41_683: EngineState: Cranking timeout 1 seconds
```

rusefillc, 2023-11-09:
proteus
```
2023-11-08_22_54_19_101: EngineState: ac_switch timeLast 126926583
2023-11-08_22_54_19_102: EngineState: physical state 1 value 1
2023-11-08_22_54_19_102: EngineState: mode 0
2023-11-08_22_54_19_102: EngineState: start_button timeLast 0
2023-11-08_22_54_19_103: EngineState: physical state 0 value 0
2023-11-08_22_54_19_105: EngineState: mode 0
2023-11-08_22_54_19_105: EngineState: confirmation_debounce:8
2023-11-08_22_54_22_013: EngineState: ac_switch timeLast 138926594
2023-11-08_22_54_22_014: EngineState: physical state 1 value 1
2023-11-08_22_54_22_015: EngineState: mode 0
2023-11-08_22_54_22_015: EngineState: start_button timeLast 0
2023-11-08_22_54_22_016: EngineState: physical state 0 value 0
2023-11-08_22_54_22_017: EngineState: mode 0
2023-11-08_22_54_22_018: EngineState: confirmation_debounce:8
2023-11-08_22_54_23_301: EngineState: ac_switch timeLast 143926553
2023-11-08_22_54_23_301: EngineState: physical state 1 value 1
2023-11-08_22_54_23_301: EngineState: mode 0
2023-11-08_22_54_23_302: EngineState: start_button timeLast 0
2023-11-08_22_54_23_302: EngineState: physical state 0 value 0
2023-11-08_22_54_23_304: EngineState: mode 0
2023-11-08_22_54_23_309: EngineState: confirmation_debounce:8
2023-11-08_22_54_42_452: EngineState: engine has STOPPED
2023-11-08_22_54_42_454: EngineState: trigger PAL mode 1
2023-11-08_22_54_42_454: EngineState: Template HYUNDAI_PB (104) trigger TT_TOOTHED_WHEEL_60_2 (8) syncEdge=RiseOnly tdcOffset=0.00
2023-11-08_22_54_42_455: EngineState: trigger#1 event counters up=0/down=1
2023-11-08_22_54_42_460: EngineState: expected cycle events 58/0
2023-11-08_22_54_42_472: EngineState: trigger type=8/need2ndChannel=No
2023-11-08_22_54_42_475: EngineState: synchronizationNeeded=Yes/isError=No/total errors=0 ord_err=0/total revolutions=0/self=No
2023-11-08_22_54_42_480: EngineState: gap from 2.25 to 3.75
2023-11-08_22_54_42_485: EngineState: primary trigger input: VR 1
2023-11-08_22_54_42_490: EngineState: primary trigger simulator: NONE OM_DEFAULT freq=1200
2023-11-08_22_54_42_493: EngineState: VVT input: Digital 1 mode VVT_SINGLE_TOOTH
2023-11-08_22_54_42_496: EngineState: VVT 0 event counters: 0/0
2023-11-08_22_54_42_499: EngineState: VVT input: Digital 2 mode VVT_SINGLE_TOOTH
2023-11-08_22_54_42_501: EngineState: VVT 1 event counters: 0/0
2023-11-08_22_54_42_503: EngineState: primary logic input: NONE
2023-11-08_22_54_42_505: EngineState: secondary logic input: NONE
2023-11-08_22_54_42_507: EngineState: totalTriggerHandlerMaxTime=195
2023-11-08_22_54_42_510: EngineState: engine sniffer current size=2
2023-11-08_22_54_42_703: EngineState: Firing priming pulse of 43.12 ms
2023-11-08_22_54_47_203: EngineState: ac_switch timeLast 239526558
2023-11-08_22_54_47_204: EngineState: physical state 1 value 1
2023-11-08_22_54_47_204: EngineState: mode 0
2023-11-08_22_54_47_204: EngineState: start_button timeLast 0
2023-11-08_22_54_47_204: EngineState: physical state 0 value 0
2023-11-08_22_54_47_206: EngineState: mode 0
2023-11-08_22_54_47_223: EngineState: confirmation_debounce:8
2023-11-08_22_54_49_005: EngineState: ac_switch timeLast 246726559
2023-11-08_22_54_49_006: EngineState: physical state 1 value 1
2023-11-08_22_54_49_006: EngineState: mode 0
2023-11-08_22_54_49_009: EngineState: start_button timeLast 0
2023-11-08_22_54_49_009: EngineState: physical state 0 value 0
2023-11-08_22_54_49_018: EngineState: mode 0
2023-11-08_22_54_49_020: EngineState: confirmation_debounce:8
2023-11-08_22_54_50_304: EngineState: ac_switch timeLast 251726587
2023-11-08_22_54_50_305: EngineState: physical state 1 value 1
2023-11-08_22_54_50_306: EngineState: mode 0
2023-11-08_22_54_50_309: EngineState: start_button timeLast 0
2023-11-08_22_54_50_326: EngineState: physical state 0 value 0
2023-11-08_22_54_50_328: EngineState: mode 0
2023-11-08_22_54_50_331: EngineState: confirmation_debounce:8
2023-11-08_22_54_51_502: EngineState: ac_switch timeLast 256526587
2023-11-08_22_54_51_502: EngineState: physical state 1 value 1
2023-11-08_22_54_51_502: EngineState: mode 0
2023-11-08_22_54_51_504: EngineState: start_button timeLast 0
2023-11-08_22_54_51_504: EngineState: physical state 0 value 0
2023-11-08_22_54_51_511: EngineState: mode 0
2023-11-08_22_54_51_512: EngineState: confirmation_debounce:8
2023-11-08_22_54_53_306: EngineState: ac_switch timeLast 263926560
2023-11-08_22_54_53_307: EngineState: physical state 1 value 1
2023-11-08_22_54_53_308: EngineState: mode 0
2023-11-08_22_54_53_309: EngineState: start_button timeLast 0
2023-11-08_22_54_53_312: EngineState: physical state 0 value 0
2023-11-08_22_54_53_327: EngineState: mode 0
2023-11-08_22_54_53_330: EngineState: confirmation_debounce:8
2023-11-08_22_54_54_605: EngineState: ac_switch timeLast 268926588
2023-11-08_22_54_54_606: EngineState: physical state 1 value 1
2023-11-08_22_54_54_609: EngineState: mode 0
2023-11-08_22_54_54_624: EngineState: start_button timeLast 0
2023-11-08_22_54_54_628: EngineState: physical state 0 value 0
2023-11-08_22_54_54_631: EngineState: mode 0
2023-11-08_22_54_54_634: EngineState: confirmation_debounce:8
2023-11-08_22_54_56_404: EngineState: ac_switch timeLast 276326559
2023-11-08_22_54_56_404: EngineState: physical state 1 value 1
2023-11-08_22_54_56_405: EngineState: mode 0
2023-11-08_22_54_56_407: EngineState: start_button timeLast 0
2023-11-08_22_54_56_416: EngineState: physical state 0 value 0
2023-11-08_22_54_56_418: EngineState: mode 0
2023-11-08_22_54_56_420: EngineState: confirmation_debounce:8
2023-11-08_22_54_57_604: EngineState: ac_switch timeLast 281126559
2023-11-08_22_54_57_604: EngineState: physical state 1 value 1
2023-11-08_22_54_57_605: EngineState: mode 0
2023-11-08_22_54_57_607: EngineState: start_button timeLast 0
2023-11-08_22_54_57_623: EngineState: physical state 0 value 0
2023-11-08_22_54_57_625: EngineState: mode 0
2023-11-08_22_54_57_627: EngineState: confirmation_debounce:8
```

rusefillc, 2023-11-09:
proteus AC pin off

```
2023-11-08_22_57_30_149: EngineState: ac_switch timeLast 0
2023-11-08_22_57_30_150: EngineState: physical state 0 value 0
2023-11-08_22_57_30_151: EngineState: mode 0
2023-11-08_22_57_30_153: EngineState: start_button timeLast 0
2023-11-08_22_57_30_162: EngineState: physical state 0 value 0
2023-11-08_22_57_30_167: EngineState: mode 0
2023-11-08_22_57_30_170: EngineState: confirmation_debounce:8
2023-11-08_22_57_31_949: EngineState: ac_switch timeLast 0
2023-11-08_22_57_31_950: EngineState: physical state 0 value 0
2023-11-08_22_57_31_950: EngineState: mode 0
2023-11-08_22_57_31_950: EngineState: start_button timeLast 0
2023-11-08_22_57_31_953: EngineState: physical state 0 value 0
2023-11-08_22_57_31_953: EngineState: mode 0
2023-11-08_22_57_31_959: EngineState: confirmation_debounce:8
2023-11-08_22_57_33_051: EngineState: ac_switch timeLast 0
2023-11-08_22_57_33_052: EngineState: physical state 0 value 0
2023-11-08_22_57_33_053: EngineState: mode 0
2023-11-08_22_57_33_057: EngineState: start_button timeLast 0
2023-11-08_22_57_33_069: EngineState: physical state 0 value 0
2023-11-08_22_57_33_073: EngineState: mode 0
2023-11-08_22_57_33_077: EngineState: confirmation_debounce:8
2023-11-08_22_57_42_901: EngineState: WARNING: Sensor fault: Tps1 is inconsistent
2023-11-08_22_57_44_552: EngineState: ac_switch timeLast 0
2023-11-08_22_57_44_553: EngineState: physical state 0 value 0
2023-11-08_22_57_44_554: EngineState: mode 0
2023-11-08_22_57_44_555: EngineState: start_button timeLast 0
2023-11-08_22_57_44_560: EngineState: physical state 0 value 0
2023-11-08_22_57_44_572: EngineState: mode 0
2023-11-08_22_57_44_576: EngineState: confirmation_debounce:8
2023-11-08_22_57_45_950: EngineState: ac_switch timeLast 0
2023-11-08_22_57_45_951: EngineState: physical state 0 value 0
2023-11-08_22_57_45_951: EngineState: mode 0
2023-11-08_22_57_45_951: EngineState: start_button timeLast 0
2023-11-08_22_57_45_953: EngineState: physical state 0 value 0
2023-11-08_22_57_45_953: EngineState: mode 0
2023-11-08_22_57_45_959: EngineState: confirmation_debounce:8
2023-11-08_22_57_47_048: EngineState: ac_switch timeLast 0
2023-11-08_22_57_47_049: EngineState: physical state 0 value 0
2023-11-08_22_57_47_049: EngineState: mode 0
2023-11-08_22_57_47_050: EngineState: start_button timeLast 0
2023-11-08_22_57_47_053: EngineState: physical state 0 value 0
2023-11-08_22_57_47_053: EngineState: mode 0
2023-11-08_22_57_47_063: EngineState: confirmation_debounce:8
```

rusefillc, 2023-11-09:
[image]

rusefillc, 2023-11-09:
[image]

rusefillc, 2023-11-09:
[image]

rusefillc, 2023-11-09:
```
2023-11-09_00_12_34_380: EngineState: start_button getPhysicalState 1
2023-11-09_00_12_34_390: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
2023-11-09_00_12_35_061: EngineState: pin C4 - 2Step/Digital value 0
2023-11-09_00_12_35_061: EngineState: confirmation_readpin pf11:12
2023-11-09_00_12_35_480: EngineState: start_button getPhysicalState 0
2023-11-09_00_12_35_481: EngineState: Cranking timeout 1 seconds
2023-11-09_00_12_35_482: EngineState: start_button getPhysicalState 1
2023-11-09_00_12_35_485: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
2023-11-09_00_12_35_962: EngineState: pin C4 - 2Step/Digital value 0
2023-11-09_00_12_35_962: EngineState: confirmation_readpin pf11:12
2023-11-09_00_12_36_466: EngineState: pin C4 - 2Step/Digital value 0
2023-11-09_00_12_36_467: EngineState: confirmation_readpin pf11:12
2023-11-09_00_12_36_564: EngineState: start_button getPhysicalState 0
2023-11-09_00_12_36_565: EngineState: Cranking timeout 1 seconds
2023-11-09_00_12_36_567: EngineState: start_button getPhysicalState 1
2023-11-09_00_12_36_568: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
2023-11-09_00_12_36_866: EngineState: pin C4 - 2Step/Digital value 0
2023-11-09_00_12_36_868: EngineState: confirmation_readpin pf11:12
2023-11-09_00_12_37_165: EngineState: pin C4 - 2Step/Digital value 0
2023-11-09_00_12_37_166: EngineState: confirmation_readpin pf11:12
2023-11-09_00_12_37_364: EngineState: pin C4 - 2Step/Digital value 0
2023-11-09_00_12_37_365: EngineState: confirmation_readpin pf11:12
2023-11-09_00_12_37_761: EngineState: start_button getPhysicalState 0
2023-11-09_00_12_37_761: EngineState: Cranking timeout 1 seconds
2023-11-09_00_12_37_762: EngineState: start_button getPhysicalState 1
2023-11-09_00_12_37_770: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
2023-11-09_00_12_37_772: EngineState: pin C4 - 2Step/Digital value 0
2023-11-09_00_12_37_775: EngineState: confirmation_readpin pf11:12
2023-11-09_00_12_37_861: EngineState: pin C4 - 2Step/Digital value 0
2023-11-09_00_12_37_861: EngineState: confirmation_readpin pf11:12
```

rusefillc, 2023-11-09:
```
    storedValue = false;
    return readPinState();
```

looks like fragile concurrency? Really concerned about too many assignments into the variable

rusefillc, 2023-11-09:
```
2023-11-09_09_04_58_558: EngineState: start_button value 0
2023-11-09_09_04_58_559: EngineState: Cranking timeout 1 seconds
2023-11-09_09_04_58_618: EngineState: start_button value 1
2023-11-09_09_04_58_619: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
2023-11-09_09_04_59_660: EngineState: start_button value 0
2023-11-09_09_04_59_693: EngineState: Cranking timeout 1 seconds
2023-11-09_09_04_59_723: EngineState: start_button value 1
2023-11-09_09_04_59_724: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
2023-11-09_09_05_00_760: EngineState: start_button value 0
2023-11-09_09_05_00_761: EngineState: Cranking timeout 1 seconds
2023-11-09_09_05_00_817: EngineState: start_button value 1
2023-11-09_09_05_00_818: EngineState: Let's crank this engine for up to 1 seconds via F1 - Idle2!
```

rusefillc, 2023-11-09:
[image]

rusefillc, 2023-11-09:
@ElDominio i think I have a HW fix

1K pull-down between GND and C4 start request button

[image]

[image]

ElDominio, 2023-11-10:
> @ElDominio i think I have a HW fix
> 
> 1K pull-down between GND and C4 start request button
> 
> [image]
> 
> [image]

Can confirm hardware fix is ok! Now, why?

rusefillc, 2024-09-15:
@ElDominio I am starting to think that this is NOT a "start button" and "starter relay"

note how ECU input and key are one side of the relay, and ECU output is the low side of same relay

it's more the case of "ECU can PROHIBIT cranking but REMOVING low side above specific RPM, and ign KEY is how you crank, and ECU has a chance to read fact that we are cranking for whatever reasons"

got kia happy with GPPWM all set to 100% unconditional low side starter relay and no ECU logic

[image]

closed 2024-09-16 by rusefillc