rusefi/rusefi #5704: Conditional GPIO (No PWM, ONLY ON/FF)

ElDominio, 2023-11-10:
**Is your feature request related to a problem? 
I like the GP PWM tables very much, but I'm always frustrated when I have to waste one of those tables to turn an output on/off.

**Would we need new configuration fields?**
New dialog boxes for if statements, where we can choose for example, if RPM/MAP greater/less than X, activate GPIO, with hysteresis as usual

**Describe the solution you'd like**
Add these options

**Describe alternatives you've considered**
Using the GPIO PWM tables works great, but many neandrathals have a lot of problems wrapping their heads around writing "100%" on a table, and besides, we only have 4 of those tables. Conditional outputs would keep these free

ElDominio, 2025-01-18:
@FDSoftware

rusefillc, 2025-04-17:
See #7735