rusefi/rusefi #5715: vtwin says fixed timing should accounts for cylinder offset

rusefillc, 2023-11-13:
cylinder offset is for real it should be accounted on fixed timing ignition mode

mck1117, 2023-11-13:
I believe cylinder offset and trim are always applied even when in fixed mode

rusefillc, 2023-11-13:
```
	// Allow correction only if set to dynamic
	// AND we're either not cranking OR allowed to correct in cranking
	bool allowCorrections = engineConfiguration->timingMode == TM_DYNAMIC
		&& (!isCranking || engineConfiguration->useAdvanceCorrectionsForCranking);

	if (allowCorrections) {
		angle_t correction = getAdvanceCorrections(engineLoad);
		if (!cisnan(correction)) { // correction could be NaN during settings update
			angle += correction;
		}
	}
```

mck1117, 2023-11-13:
And cylinder offset isn't related to that function:

```c++
static angle_t getAdvanceCorrections(float engineLoad) {

	.....

	return engine->ignitionState.timingIatCorrection
		+ engine->ignitionState.cltTimingCorrection
		+ engine->ignitionState.timingPidCorrection;
}
```

rusefillc, 2023-11-13:
@mck1117 what I with timing light see seems worse - looks like first two revolutions there is no offset on second cylinder and then it appears. that's in fixed timing mode, very repetitive.

rusefillc, 2023-11-13:
[image]

rusefillc, 2023-11-13:
https://rusefi.com/online/view.php?log=1320

rusefillc, 2023-11-13:
[image]

mck1117, 2023-11-13:
Note uneven spacing of both fuel and spark while in fixed mode.

[image]

Probably has to do with this part: `at least on FOME they are`

commit f50474b0 (2023-11-13): odd fire engines do not appreciate wasted mode on spinning fix #5715
closed 2023-11-13 by rusefillc