rusefi/rusefi #5716: alternator refactoring open questions

rusefillc, 2023-11-13:
https://github.com/rusefi/rusefi/commit/3e9b2d0e1f87f4b6990e4a16f81b1b6c9ca4e2a7#r132373139 and https://github.com/rusefi/rusefi/commit/3e9b2d0e1f87f4b6990e4a16f81b1b6c9ca4e2a7#r132373388 should be addressed

andreika-git, 2023-11-13:
We are passing parameter by reference, so until we start using PID, everything is fine. We have time to read the config and fill in the PID settings before we start using it.

mck1117, 2023-11-13:
> We are passing parameter by reference, so until we start using PID, everything is fine. We have time to read the config and fill in the PID settings before we start using it.

That doesn't mean it isn't UB. You can't guarantee the behavior of the `->` operator until the object has been fully initialized.

mck1117, 2023-11-13:
In fact - the current implementation even relies on the static constructor order. It's not required that `engineConfiguration` be optimized out by the compiler, which means it may be null until all static construction is complete. It so happens that the compiler _does_ make that optimization, but it's not a guarantee.

andreika-git, 2023-11-13:
> That doesn't mean it isn't UB. You can't guarantee the behavior of the `->` operator until the object has been fully initialized.

True, that's not guaranteed. The problem existed before the refactoring. We should either move it to init() called from initAlternatorCtrl(), or pass &persistentState.persistentConfiguration (but there's also EFI_UNIT_TEST so it's more complicated).

andreika-git, 2023-11-13:
https://github.com/rusefi/rusefi/pull/5717