rusefi/rusefi #5719: WBO firmware update

rusefillc, 2023-11-14:
what if one has a board with two WBO modules on board? how is ```setWidebandOffset``` supposed to work if it ```Setting all connected widebands to index %d...```?

what about updateWidebandFirmware? how does one update only one unit?

mck1117, 2023-11-15:
> one has a board with two WBO modules on board

this is what https://github.com/mck1117/wideband/issues/11 is for, no?

rusefillc, 2023-11-15:
> 
> this is what [mck1117/wideband#11](https://github.com/mck1117/wideband/issues/11) is for, no?

yes individual units know their own indices thanks to https://github.com/mck1117/wideband/issues/11

but it does not seem like current bootloader custom protocol has provisions to talk only to specified controlled by index as of today? while perfect ideal solution would be to expand protocol it's tempting to just make it work by hanging the other one. I assume that hanging up one of controllers would be way less effort and much less coordination

rusefillc, 2024-03-05:
@dron0gus now has nissan121 as test hw

[image]