rusefi/rusefi #5723: Test if all critical pins are really off when fatal_error or hard_fault occurs

andreika-git, 2023-11-14:


rusefillc, 2023-11-14:
two years ago we had
```
	addConsoleAction("error", myerror);
static void myerror(void) {
	firmwareError(CUSTOM_ERR_TEST_ERROR, "firmwareError: %d", getRusEfiVersion());
}
```

andreika-git, 2023-11-15:
Test results:

1) Setup: Discovery-F407, firmware master.2023.11.15.f407-discovery.3499919882, measuring voltage on PE14 (IGN1)

2) Test scenario 1:
  * "enable_self_stimulation": IGN1 = 0..3.0..0..3.0.. (changing)
  * "error": IGN1 = 0

3) Test scenario 2:
  * "enable_self_stimulation": IGN1 = 0..3.0..0..3.0.. (changing)
  * "hard_fault": IGN1 = 0

To be absolutely sure, I modified the firmware to comment out setLow() in spartLogic.cpp

4) Test scenario 3:
  * "enable_self_stimulation": IGN1 = 3.0
  * "error": IGN1 = 0

4) Test scenario 4:
  * "enable_self_stimulation": IGN1 = 3.0
  * "hard_fault": IGN1 = 0  

5) Test scenario 5:
  * "enable_self_stimulation": IGN1 = 3.0
  * "disable_self_stimulation": IGN1 = 0  

It seems that we can be certain enough that the ignition pins are off when a fatal error or hard fault occurs.

rusefillc, 2023-11-16:
> To be absolutely sure, I modified the firmware to comment out setLow() in spartLogic.cpp

please push this let's have an official custom branch so that we can do similar manual test later on.

andreika-git, 2023-11-16:
https://github.com/andreika-git/rusefi/tree/test_ignition_pins_when_fatal_occurs

closed 2023-11-16 by rusefillc