rusefi/rusefi #5742: Lua: setAcDisabled is broken, only disabled, never re-enables

ElDominio, 2023-11-21:
**Describe the bug**
Calling setAcDisabled disables AC, doesn't matter the value passed (either 0 or 1)

**Hardware**
hyundai154

**To Reproduce**
Write setAcDisabled(0) in lua
ac output never activates if this function is called

**Expected behavior**
setAcDisabled(0) allows normal activation of AC, 1 disables AC oupurt

rusefillc, 2023-11-21:
I am able to reproduce the problem I know what it could be about

[image]

[image]

rusefillc, 2023-11-21:
@ElDominio Lua is fucked about about number to boolean conversion

1 is true and 0 is true :(

only ``nil`` is false.

[image]

ElDominio, 2023-11-22:
So use nil instead of 0?

rusefillc, 2023-11-22:
> So use nil instead of 0?

yes, at least for now. we kind of _can_ handle boolean in a not-so-Lua way but that's a can of warms to open

ElDominio, 2023-11-22:
I'm fine with using nil, not that hard to remember

ElDominio, 2024-02-23:
The problem was that i did not use (nil) and expected (0) to enabe AC. This can be closed

closed 2024-02-23 by rusefillc