rusefi/rusefi #5755: MAP Prediction for accel enrichment

ElDominio, 2023-11-26:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I have to set up acce enrich and hunt values that "work". 
Using MAP prediction, we could simplify transient fueling by using the already-existing MAP prediction table, and blending it into the main fuel model in transient events (where MAP is working correcty, meaning we aren't using the MAP prediction table to run the car)

**Would we need new configuration fields?**
We'd only need a value for MAP Prediction-to-SD blend time vs RPM (2D)

**Describe the solution you'd like**
I'd like to be able to populate the MAP prediction table, and use it for transient events instead of using our current accel enrich implementation.

**Additional context**
[image]

https://youtu.be/u4mtOfWiyZo

Very long video but very informative

highspeed427, 2024-06-21:
Keep current Baro Pressure in mind for this setup.

ElDominio, 2024-06-24:
Informational video: https://youtu.be/a2oq4GeTvdQ

ElDominio, 2024-12-15:
https://youtu.be/jJ3TiaQTzYQ?t=102

exact explanation

ElDominio, 2025-01-18:
@FDSoftware  esta seria la tesis doctoral

ElDominio, 2025-07-01:
[image]

will be testing this on my branch soon

ElDominio, 2025-07-02:
[image]

@rusefillc my friend is trying to make useful code. i will test his useful code somewhat soon

ElDominio, 2026-02-23:
I think this exists