rusefi/rusefi #5759: Lua VSS broken for no apparent reson?

ElDominio, 2023-11-28:
**Describe the bug**
2023-11-28_18_19_01_632: EngineState: LUA ERROR loading script: [string "..."]:7: Tried to create a Lua sensor of type VehicleSpeed, but one was already registered.

It was working fine, after a write it started throwing this error

Lua code here:
https://gist.github.com/ElDominio/1b2c2329c52d7a68924c65014bd46bd2

**Hardware**
 - 4chan
 - 3a91

**To Reproduce**
Steps to reproduce the behavior:
Write Lua code with vss call, sometimes is broken/?

**Expected behavior**
I changed nothing in the code and after a burn Lua complained about bad vss

**Screenshots**
[image]

ElDominio, 2023-11-28:
Rebooting ECU reverted the problem?
Same error again on Nov19 firmware

rusefillc, 2023-11-28:
@ElDominio here is minimal script and it works for me

```

speedSensor = Sensor.new("VehicleSpeed")
speedSensor : set(123)

print("Speed is " ..getSensor("VehicleSpeed"))

function onTick()
end

```

@ElDominio please report output of this minimal script on your end

ElDominio, 2023-11-28:
2023-11-28_19_04_44_609: EngineState: LUA ERROR loading script: [string "speedSensor = Sensor.new("VehicleSpeed")..."]:1: Tried to create a Lua sensor of type VehicleSpeed, but one was already registered.

ElDominio, 2023-11-28:
changed firmware to latest official release, added Lua and ECU is happy

rusefillc, 2023-11-29:
@ElDominio looks like the whole "can vss: lua" is totally unneeded?

ElDominio, 2023-12-01:
CAN VSS should be under the CAN settings, built into cars with CANs that we have decoded

But what is someone hss physical VSS, and they inject through Lua too, will Lua console notice and scream?

ElDominio, 2023-12-08:
I dunno if you want this thread, but somution is to disabled CAN LUA VSS in TS and inject it anyways

commit 6b3c82e8 (2023-12-12): only: Lua VSS broken for no apparent reson? fix #5759
closed 2023-12-12 by rusefillc