rusefi/rusefi #5765: diag ignition using 33810

rusefillc, 2023-12-02:
we shall run a bench test of ignition test on 33810

dron0gus, 2023-12-02:
```
diff --git a/firmware/hw_layer/drivers/gpio/mc33810.cpp b/firmware/hw_layer/drivers/gpio/mc33810.cpp
index b5fe8eb032..86426fb69a 100644
--- a/firmware/hw_layer/drivers/gpio/mc33810.cpp
+++ b/firmware/hw_layer/drivers/gpio/mc33810.cpp
@@ -351,9 +351,8 @@ int Mc33810::chip_init()
                        goto err_gpios;
                }
 
-               /* set IGN/GP outputs to GP mode - to be fixed!
-                * disable retry after recovering from under/overvoltage */
-               ret = spi_rw(MC_CMD_MODE_SELECT((0xf << 8) | (1 << 6)) , NULL);
+               /* disable retry after recovering from under/overvoltage */
+               ret = spi_rw(MC_CMD_MODE_SELECT((1 << 6)) , NULL);
                if (ret) {
                    efiPrintf(DRIVER_NAME " cmd mode select");
                        goto err_gpios;
```

rusefillc, 2024-01-03:
https://github.com/rusefi/rusefi/issues/5860

rusefillc, 2024-09-20:
@dron0gus has fancy 154hyndai for tests now

closed 2025-05-16 by rusefillc