rusefi/rusefi #5773: two 042 modules sporadic reading

rusefillc, 2023-12-05:
proteus and adapter board with two 042 WBO modules

first module produces reasonable line second module value is sporadic. so far we've tried flipping plugs - the issue stays with 2nd module both sensors are good as long as hooked to first module

[image]

[image]
[DataLogs12-5_LF1.zip](https://github.com/rusefi/rusefi/files/13573462/DataLogs12-5_LF1.zip)

rusefillc, 2023-12-06:
https://github.com/mck1117/wideband/pull/302

rusefillc, 2023-12-06:
At the moment my best random guess is that second WBO has troubles communicating due to bus load but maybe timing causing sensor timeout? I know it's a bit far fetched but both WBO units start at exactly the same time and they TX with same constant period so they are bound to be conflicting. Technically an easy fix would be to add a bit of randomness but first this theory needs to be proven

here's an interesting fact supporting my guess: on a different board with two WBO modules and zero CAN traffic: with zero load I've received 0.1% more packets from wbo0 than from wbo1

```
2023-12-05_22_52_04_732: EngineState: LUA: 473917 473558
2023-12-05_22_52_05_721: EngineState: LUA: 474015 473656
2023-12-05_22_52_06_710: EngineState: LUA: 474113 473754
2023-12-05_22_52_07_690: EngineState: LUA: 474212 473853
2023-12-05_22_52_08_671: EngineState: LUA: 474310 473951
```

```
setTickRate(300)

local wbo0counter = 0
local wbo1counter = 0
function onWbo0(bus, id, dlc, data)
	wbo0counter = wbo0counter + 1
end

function onWbo1(bus, id, dlc, data)
	wbo1counter = wbo1counter + 1
end

everySecond = Timer.new()

canRxAdd(1, 0x190, onWbo0)
canRxAdd(1, 0x192, onWbo1)

function onTick()
	if (everySecond : getElapsedSeconds() > 1) then
		everySecond : reset()
		print(wbo0counter .." " ..wbo1counter)
	end
end
```

in the meantime there is also https://github.com/rusefi/wideband/commit/5f45c92521fb7178fa4224b6972b1ddacd0c1c66 to display TX error condition via LEDs

@mck1117 @dron0gus

mck1117, 2023-12-06:
why don't you scope the bus instead of speculating about bus load

two WBOs transmitting is a tiny tiny TINY fraction of a bus worth of bandwidth

rusefillc, 2023-12-06:
> why don't you scope the bus instead of speculating about bus load

great question! I've failed for reasons unclear to me. I've attempted with PCAN/pcan view and something weird was happening: rusEFI verbosecan was showing 0x190 and 192 while PCANview was showing lots of CANbus activity but nothing 0x19x.

mck1117, 2023-12-06:
> I've received 0.1% more packets from wbo0 than from wbo1

this is well within the oscillator tolerance on the wideband MCU

benassi124, 2023-12-06:
on real vehicle

```
2023-12-06_08_44_49_369: EngineState: LUA: 4031 4032
2023-12-06_08_44_50_369: EngineState: LUA: 4129 4130
2023-12-06_08_44_51_272: EngineState: LUA: 4227 4228
2023-12-06_08_44_52_298: EngineState: LUA: 4325 4326
2023-12-06_08_44_53_285: EngineState: LUA: 4423 4423
2023-12-06_08_44_54_291: EngineState: LUA: 4521 4521
```

benassi124, 2023-12-06:
```
2023-12-06_13_55_04_469: EngineState: LUA: from wbo1 402  A0 0 64 25 7 3 0 0
2023-12-06_13_55_04_469: EngineState: LUA: from wbo0 400  A0 1 AD 27 B 3 0 0
2023-12-06_13_55_04_469: EngineState: LUA: from wbo1 402  A0 0 15 2B 8 3 0 0
2023-12-06_13_55_04_469: EngineState: LUA: from wbo0 400  A0 1 9F 27 B 3 0 0
2023-12-06_13_55_04_470: EngineState: LUA: from wbo1 402  A0 1 4 25 6 3 0 0
2023-12-06_13_55_04_472: EngineState: LUA: from wbo0 400  A0 1 6E 27 B 3 0 0
2023-12-06_13_55_04_477: EngineState: LUA: from wbo1 402  A0 0 B8 2B 8 3 0 0
2023-12-06_13_55_04_478: EngineState: LUA: from wbo0 400  A0 1 7D 27 B 3 0 0
2023-12-06_13_55_04_480: EngineState: LUA: from wbo1 402  A0 1 6A 25 8 3 0 0
2023-12-06_13_55_04_481: EngineState: LUA: from wbo0 400  A0 1 4A 27 B 3 0 0
2023-12-06_13_55_04_482: EngineState: LUA: from wbo1 402  A0 0 EC 29 A 3 0 0
```

benassi124, 2023-12-06:
```
2023-12-06_14_05_53_858: EngineState: LUA: from wbo1 402  A0 0 97 25 9 3 0 0
2023-12-06_14_05_53_858: EngineState: LUA: from wbo0 400  A0 1 B8 2B B 3 0 0
2023-12-06_14_05_53_858: EngineState: LUA: from wbo1 402  A0 1 8F 22 A 3 0 0
2023-12-06_14_05_53_858: EngineState: LUA: from wbo0 400  A0 1 7A 2C B 3 0 0
2023-12-06_14_05_53_858: EngineState: LUA: from wbo1 402  A0 0 B6 25 C 3 0 0
2023-12-06_14_05_53_859: EngineState: LUA: from wbo0 400  A0 1 63 2C B 3 0 0
2023-12-06_14_05_53_956: EngineState: LUA: from wbo0 400  A0 1 BB 2D B 3 0 0
2023-12-06_14_05_53_956: EngineState: LUA: from wbo1 402  A0 0 66 26 A 3 0 0
2023-12-06_14_05_53_956: EngineState: LUA: from wbo0 400  A0 1 EF 2D B 3 0 0
2023-12-06_14_05_53_957: EngineState: LUA: from wbo1 402  A0 0 82 23 9 3 0 0
2023-12-06_14_05_53_957: EngineState: LUA: from wbo0 400  A0 1 A3 2E B 3 0 0
2023-12-06_14_05_53_957: EngineState: LUA: from wbo1 402  A0 0 37 27 B 3 0 0
2023-12-06_14_05_53_957: EngineState: LUA: from wbo0 400  A0 1 E5 2E B 3 0 0
2023-12-06_14_05_53_957: EngineState: LUA: from wbo1 402  A0 1 4F 23 8 3 0 0
2023-12-06_14_05_53_957: EngineState: LUA: from wbo0 400  A0 1 13 2E B 3 0 0
2023-12-06_14_05_53_957: EngineState: LUA: from wbo1 402  A0 0 9C 27 9 3 0 0
2023-12-06_14_05_53_957: EngineState: LUA: from wbo0 400  A0 1 11 2C B 3 0 0
2023-12-06_14_05_53_957: EngineState: LUA: from wbo1 402  A0 1 95 23 9 3 0 0
2023-12-06_14_05_53_957: EngineState: LUA: from wbo0 400  A0 1 FD 2A B 3 0 0
2023-12-06_14_05_53_957: EngineState: LUA: from wbo0 400  A0 1 11 2A B 3 0 0
2023-12-06_14_05_53_958: EngineState: LUA: from wbo1 402  A0 0 ED 23 9 3 0 0
2023-12-06_14_05_53_958: EngineState: LUA: from wbo0 400  A0 1 15 2A B 3 0 0
2023-12-06_14_05_53_958: EngineState: LUA: from wbo1 402  A0 0 DE 27 9 3 0 0
2023-12-06_14_05_54_057: EngineState: LUA: from wbo0 400  A0 1 41 2A B 3 0 0
2023-12-06_14_05_54_057: EngineState: LUA: from wbo1 402  A0 0 CF 23 7 3 0 0
2023-12-06_14_05_54_057: EngineState: LUA: from wbo0 400  A0 1 A6 29 B 3 0 0
2023-12-06_14_05_54_057: EngineState: LUA: from wbo1 402  A0 0 79 28 9 3 0 0
2023-12-06_14_05_54_057: EngineState: LUA: from wbo0 400  A0 1 E8 29 B 3 0 0
2023-12-06_14_05_54_057: EngineState: LUA: from wbo1 402  A0 0 97 24 9 3 0 0
2023-12-06_14_05_54_058: EngineState: LUA: from wbo0 400  A0 1 7C 29 B 3 0 0
2023-12-06_14_05_54_058: EngineState: LUA: from wbo1 402  A0 0 C5 29 9 3 0 0
2023-12-06_14_05_54_058: EngineState: LUA: from wbo0 400  A0 1 8F 29 B 3 0 0
2023-12-06_14_05_54_058: EngineState: LUA: from wbo0 400  A0 1 81 29 B 3 0 0
2023-12-06_14_05_54_058: EngineState: LUA: from wbo1 402  A0 0 18 29 9 3 0 0
2023-12-06_14_05_54_058: EngineState: LUA: from wbo0 400  A0 1 76 29 B 3 0 0
2023-12-06_14_05_54_058: EngineState: LUA: from wbo1 402  A0 0 F6 24 8 3 0 0
2023-12-06_14_05_54_058: EngineState: LUA: from wbo0 400  A0 1 CA 28 B 3 0 0
2023-12-06_14_05_54_058: EngineState: LUA: from wbo1 402  A0 0 EA 29 8 3 0 0
2023-12-06_14_05_54_058: EngineState: LUA: from wbo0 400  A0 1 DC 28 B 3 0 0
2023-12-06_14_05_54_059: EngineState: LUA: from wbo1 402  A0 0 19 25 5 3 0 0
2023-12-06_14_05_54_059: EngineState: LUA: from wbo0 400  A0 1 D8 28 B 3 0 0
```

benassi124, 2023-12-06:
```
2023-12-06_14_08_31_677: EngineState: LUA: from wbo0 diag 401  2B 1 C1 1 84 0 99 0
2023-12-06_14_08_31_682: EngineState: LUA: from wbo1 diag 403  34 1 6E 4 37 0 9D 0
2023-12-06_14_08_31_683: EngineState: LUA: from wbo0 diag 401  2C 1 C1 1 87 0 99 0
2023-12-06_14_08_31_686: EngineState: LUA: from wbo1 diag 403  31 1 A7 1 FF 0 9D 0
2023-12-06_14_08_31_687: EngineState: LUA: from wbo0 diag 401  2C 1 C0 1 84 0 99 0
2023-12-06_14_08_31_691: EngineState: LUA: from wbo1 diag 403  32 1 7E 3 32 0 9D 0
2023-12-06_14_08_31_693: EngineState: LUA: from wbo0 diag 401  2B 1 C2 1 85 0 97 0
2023-12-06_14_08_31_694: EngineState: LUA: from wbo0 diag 401  2B 1 C1 1 82 0 97 0
2023-12-06_14_08_31_695: EngineState: LUA: from wbo1 diag 403  37 1 53 1 2A 0 C6 0
2023-12-06_14_08_31_778: EngineState: LUA: from wbo0 diag 401  2B 1 C1 1 83 0 97 0
2023-12-06_14_08_31_778: EngineState: LUA: from wbo0 diag 401  2B 1 C2 1 84 0 97 0
2023-12-06_14_08_31_778: EngineState: LUA: from wbo1 diag 403  39 1 9 1 33 0 C6 0
2023-12-06_14_08_31_778: EngineState: LUA: from wbo0 diag 401  2B 1 C4 1 85 0 94 0
2023-12-06_14_08_31_779: EngineState: LUA: from wbo1 diag 403  34 1 79 4 FF 0 C6 0
2023-12-06_14_08_31_779: EngineState: LUA: from wbo0 diag 401  2B 1 BF 1 81 0 94 0
2023-12-06_14_08_31_779: EngineState: LUA: from wbo0 diag 401  2B 1 C1 1 81 0 94 0
2023-12-06_14_08_31_779: EngineState: LUA: from wbo1 diag 403  32 1 6E 4 FF 0 C6 0
2023-12-06_14_08_31_779: EngineState: LUA: from wbo0 diag 401  2B 1 C2 1 85 0 94 0
2023-12-06_14_08_31_779: EngineState: LUA: from wbo1 diag 403  35 1 4B 1 2C 0 C6 0
2023-12-06_14_08_31_779: EngineState: LUA: from wbo0 diag 401  2B 1 C1 1 88 0 94 0
2023-12-06_14_08_31_779: EngineState: LUA: from wbo1 diag 403  36 1 5A 4 FF 0 C6 0
2023-12-06_14_08_31_779: EngineState: LUA: from wbo0 diag 401  2B 1 C2 1 85 0 8D 0
2023-12-06_14_08_31_779: EngineState: LUA: from wbo0 diag 401  2B 1 C1 1 86 0 8D 0
2023-12-06_14_08_31_779: EngineState: LUA: from wbo1 diag 403  35 1 78 4 FF 0 C6 0
2023-12-06_14_08_31_780: EngineState: LUA: from wbo0 diag 401  2B 1 C2 1 84 0 8D 0
2023-12-06_14_08_31_878: EngineState: LUA: from wbo0 diag 401  2B 1 C1 1 83 0 8D 0
2023-12-06_14_08_31_878: EngineState: LUA: from wbo1 diag 403  31 1 C4 4 FF 0 C9 0
2023-12-06_14_08_31_878: EngineState: LUA: from wbo0 diag 401  2B 1 C2 1 87 0 8D 0
2023-12-06_14_08_31_878: EngineState: LUA: from wbo1 diag 403  38 1 55 1 36 0 C9 0
2023-12-06_14_08_31_878: EngineState: LUA: from wbo0 diag 401  2B 1 C2 1 87 0 9A 0
2023-12-06_14_08_31_879: EngineState: LUA: from wbo1 diag 403  38 1 BA 4 FF 0 C9 0
2023-12-06_14_08_31_879: EngineState: LUA: from wbo0 diag 401  2B 1 C3 1 87 0 9A 0
2023-12-06_14_08_31_879: EngineState: LUA: from wbo0 diag 401  2B 1 C0 1 85 0 9A 0
2023-12-06_14_08_31_879: EngineState: LUA: from wbo1 diag 403  3A 1 6E 4 FF 0 C9 0
2023-12-06_14_08_31_879: EngineState: LUA: from wbo0 diag 401  2B 1 C2 1 8B 0 9A 0
2023-12-06_14_08_31_879: EngineState: LUA: from wbo1 diag 403  3C 1 45 1 33 0 C9 0
2023-12-06_14_08_31_879: EngineState: LUA: from wbo0 diag 401  2B 1 C1 1 87 0 9A 0
2023-12-06_14_08_31_879: EngineState: LUA: from wbo0 diag 401  2B 1 C1 1 87 0 94 0
2023-12-06_14_08_31_879: EngineState: LUA: from wbo1 diag 403  39 1 81 1 33 0 C9 0
2023-12-06_14_08_31_879: EngineState: LUA: from wbo0 diag 401  2B 1 C2 1 89 0 94 0
2023-12-06_14_08_31_879: EngineState: LUA: from wbo1 diag 403  38 1 99 4 FF 0 C6 0
```

rusefillc, 2023-12-06:
It was a loose RCal pin

Working on logging second wbo data on RE side

closed 2023-12-06 by rusefillc