rusefi/rusefi #5795: LuaGauge already registered?

ElDominio, 2023-12-14:
**Describe the bug**

I am trying to send technically a value from 0-2 from Lua to firmware to be able to use bias table versus said gauge. I get the following error
2023-12-14_17_20_30_299: EngineState: LUA ERROR loading script: [string "canRxAdd(0x350)..."]:12: Tried to create a Lua sensor of type LuaGauge1, but one was already registered.

**Hardware**
 -Hell154hyundai
 - G4kf

**To Reproduce**
Steps to reproduce the behavior:
Assign a ( tcStatefinal = Sensor.new("LuaGauge1") )
See error

**Expected behavior**
Sensor is passed from Lua to firmware

rusefillc, 2023-12-14:
@ElDominio please reset ECU can you reproduce the issue? I wonder if that's result of previous version of script having some error.

rusefillc, 2023-12-14:
@ElDominio proper protocol is a complete minimal script which shows issue

Can you reproduce issue say with a four line script?

rusefillc, 2023-12-14:
@ElDominio I've just improved documentation https://github.com/rusefi/rusefi_documentation/commit/eaebd7766714651103a339a922e0de6ede67f71c please try that weird inconsistent approach instead of your reasonable assumption of how this should work