rusefi/rusefi #5813: new board maybe not configured right seems to be hanging in mc33810 driver?

rusefillc, 2023-12-18:
there is no timeout on spi_lld_polled_exchange is there?

```
Thread #1 (Suspended : Signal : SIGINT:Interrupt)	
	spi_lld_polled_exchange() at hal_spi_lld.c:712 0x23e4e2	
	Mc33810::spi_rw() at mc33810.cpp:163 0x22db8e	
	Mc33810::chip_init() at mc33810.cpp:307 0x22ddda	
	Mc33810::init() at mc33810.cpp:530 0x22ddda	
	gpiochips_init() at core.cpp:231 0x238040	
	initSmartGpio() at smart_gpio.cpp:299 0x238040	
	initHardware() at hardware.cpp:514 0x238040	
	initHardware() at hardware.cpp:480 0x238040	
	commonEarlyInit() at engine_controller.cpp:660 0x238040	
	runRusEfiWithConfig() at rusefi.cpp:252 0x238040	
	runRusEfi() at rusefi.cpp:235 0x23aa24	
	main() at main.cpp:32 0x23aa24	
```

[image]

mck1117, 2023-12-18:
spipolledexchange shouldn't be able to time out unless the peripheral is configured incorrectly

rusefillc, 2023-12-18:
the peripheral was configured incorrectly

closed 2023-12-18 by rusefillc