rusefi/rusefi #5835: F407-Discovery "Maximum scheduling run length exceeded"

ElDominio, 2023-12-29:
**Describe the bug**
The error in the screenshot is thrown after cycling the pump prime a few times.  It seems to be electrical noise related since SIMILAR hardware has been used on other vehicles without this problem. I will try to get a video of what happens

**Hardware**
 - Board: Paralela Plug and Play 6G7x
 - Engine info 6G72

**To Reproduce**
Steps to reproduce the behavior:
1.  Start engine
2. Have engine stall for some reason (could be electrical noise, unclear on why)
3. Fuel pump cycles priming state for a few times
4. Processor hangs

**Expected behavior**
Only cycle once and not hang

**Screenshots**
[image]

rusefillc, 2023-12-29:
very interesting! Definitely need logs @ElDominio

ElDominio, 2023-12-29:
[image]

This is the weird thing I see in logs

At about 7.8s triggersyncratio goes crazy, but triggerchannel1 and primaryRise seem ok?

[oldprocessorlogs.zip](https://github.com/rusefi/rusefi/files/13796388/oldprocessorlogs.zip)

This log when RPM goes to 25k, it's starting to do the cycling symptom, but revising the log showed my what i mention about 7.8 that seems very strange

rusefillc, 2023-12-29:
sudden 25K RPM

[image]

ElDominio, 2024-02-23:
This is due to bad hardware. I improved hardware and get no noise which causes this... but why does noise cause software fault?

ElDominio, 2024-04-06:
No one seems to care so closing