rusefi/rusefi #5847: Change name from "invert "input": true/false" to "trigger edge: rising/falling"

ElDominio, 2024-01-02:
**Is your feature request related to a problem? Please describe.**
Current nomenclature is unclear, what is the current trigger mode? For people that are going to touch this setting, they will either invert it anyways (since they don't know what it does, just that it might work), or they might fully understand this and choose the correct edge (rising/falling) 

**Would we need new configuration fields?**
[image]

**Describe the solution you'd like**
Same as picture

**Describe alternatives you've considered**
If we are told "default edge is rising" then inverting it would logically be falling. Having text that says this would be useful

rusefillc, 2024-01-02:
[image]

closed 2024-01-02 by rusefillc

mck1117, 2024-01-02:
that's a breaking change for all tunes. Proceed with caution.