rusefi/rusefi #5873: Every time i burn new Lua, my CLT multiplier for fuel changes.

ElDominio, 2024-01-11:
**Describe the bug**
Every time I go into console and write a new Lua, a few of the temperature values on the tune change to 0.

**Hardware**
 - Board: 4chan revH

**To Reproduce**
Steps to reproduce the behavior:
Have a ready tune
Write Lua in console
Hit burn
after burn, disconnect console and connect TS
error is in screenshot

**Expected behavior**
Lua does not change CLT multiplier

**Screenshots**
[image]

**Additional context**
Tune at: 
[CurrecavaliermaxxecuntTune.zip](https://github.com/rusefi/rusefi/files/13909694/CurrecavaliermaxxecuntTune.zip)

rusefillc, 2024-01-12:
@ElDominio would you bet your kidney that rusEFI console and firmware versions are definitely the same?

ElDominio, 2024-01-12:
[image]

Today's downloads, but date modified on windows shows 12/12/2023

am i to run autoupdate even if i downloaded most recent snapshot?

rusefillc, 2024-01-12:
> am i to run autoupdate even if i downloaded most recent snapshot?

no

ElDominio, 2024-01-17:
Some odd incompatibility was culprit

rusefillc, 2024-01-21:
[image]

ElDominio, 2024-01-21:
It was probably a specific day's build. I did Lua yesterday and nothing of the sort happened

rusefillc, 2024-02-03:
We need a solution :(