rusefi/rusefi #5874: Keying on while connected through USB causes complete hang (8chan revB)

ElDominio, 2024-01-12:
**Describe the bug**
I have a car with an 8chan wired up. If I control the main relay pin (meaning the main relay pin is assigned to Main Relay output), i connect usb, and I key on, the ECU hangs.

**Hardware**
 - Board: 8chan revB
 - Engine info Ford Voodoo

**To Reproduce**
Steps to reproduce the behavior:
1. car/ecu is off
2. connect usb
3. wire Key12v to key12v input 24b and main relay output as Main Relay 11d
4. reboot ecu
5. key on (press push start button without clutch, we only want ignition voltage)
6. as soon as main relay clicks, ECU hangs

**Expected behavior**
ECU doesn't hang

**Additional context**
If main relay is removed and car is keyed on, the ECU DOES NOT crash. Also, if car is keyed on and main relay is connected afterwards, ECU DOES crash. (LED does not blink, stays on/off whatever state it was)

rusefillc, 2024-01-12:
this evokes memories of https://github.com/rusefi/rusefi/issues/4310

ElDominio, 2024-01-17:
USB IS NOT CULPRIT!

Tried again today, as soon as car is keyed on, ECU crashes. No laptop was in the vicinity of ECU

ElDominio, 2024-05-25:
It is DEIFNITELY a specific 8chan I have on hand. Swapping for another fixed this issue, would you like me to mail this ECU out?