rusefi/rusefi #5875: Disable fans after VSS > x

ElDominio, 2024-01-12:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when fans stay on at speed despite there being no reason to

**Would we need new configuration fields?**
VSS threshold for fans = after this VS, fans are disabled (since airspeed is higher than what fan motor can do anyways)

**Describe the solution you'd like**
An option as described above to disable the fans after a certain VSS

**Describe alternatives you've considered**
writing it in lua

rusefillc, 2024-01-13:
> I'm always frustrated when fans stay on at speed despite there being no reason to

Why do you care? :)

ElDominio, 2024-01-13:
[image]

> Why do you care? :)

Because we are wasting power driving them/motors die from overuse

commit 0df1a9ca (2024-01-23): only:Disable fans after VSS > x fix #5875
closed 2024-01-23 by rusefillc