rusefi/rusefi #5876: Lamborghini Diablo 1995 CKP Pattern

ElDominio, 2024-01-14:
Odd crank pattern, uses VR sensors on both crank and cam. Will try to get logic analyzer but i do need to wire up a VR converter to be able to analyze from what i think?

Ckp:
[image]

Cmp:
[image]

rusefillc, 2024-01-15:
@ElDominio from common sense I would assume google to have at least one photo of that weirdness?

See https://www.ebay.com/itm/325732593310 which at least show the six dowels matching the drawing

ElDominio, 2024-01-15:
Crank is 6/0 symmetrical VR
Cam uses two VR sensors for one tooth, since it uses two IDENTICAL ecus, one for each bank.

The two cam sensors change the phase of each bank. You can literally swap the ECUs and the car still runs

commit f9071070 (2024-01-16): only: 6/0 CKP trigger pattern for 1995 Lamborghini Diablo fix #5876
closed 2024-01-16 by rusefillc