rusefi/rusefi #5882: potting boards at point of fabrication - initial programming via main connector?

rusefillc, 2024-01-19:
some level of water proofness is desired

we have calibration via CAN and firmware _update_ via CAN

how do we use main connector even for initial programming? would we have to expose BOOT pin and use specific CAN bus with stm32 bootloader?

rusefillc, 2024-01-19:
https://www.st.com/resource/en/application_note/an2606-stm32-microcontroller-system-memory-boot-mode-stmicroelectronics.pdf

andreika-git, 2024-03-03:
I've tested the possibility to flash firmware via ST Bootloader on CAN2 using STLinkv3 as it is useful to be able to flash blank chips from the factory via CAN.

TLDR; Interim results are disappointing. So far I haven't been able to get it to work...

Setup:
- 2chan rev.F (mega-mcu144)
[image]

- STLink-V3MODS
 [image]
   * The pinout is:
  Pin 8: GND
  Pin 9: Bridge CAN RX (Input)
  Pin 10: Bridge CAN TX (Output)
  Pin 22: 5V (used for CAN transceiver module power supply)
   * See https://www.st.com/resource/en/user_manual/um2502-stlinkv3mods-and-stlinkv3mini-debuggerprogrammer-tiny-probes-for-stm32-microcontrollers-stmicroelectronics.pdf )

- CAN transceiver module based on MCP2551
[image]

- STM32CubeProgrammer v2.15.0 (Win64)

Since the bootloader uses only specific pins PB5 (Rx) and PB13 (Tx) for CAN2, and on 2chan we use pins PB12 (SPI2_CS) PB13 (SPI2_SCK), I had to modify the 2chan board by cutting the CAN-RX trace and connecting it with wire to SPI1_MOSI (PB5). The CAN-Tx was already connected correctly. Also, the transceiver in the CAN2 module was not soldered on the board - I had to solder it.

[image]
[image]
[image]

The settings for the STM32CubeProgrammer were used by default, with the exception of the bitrate, which according to AN3154 should be 125kbps.

[image]

The data is sent via the CAN bus to the transceiver and to the STM32 (checked with an oscilloscope on pin PB5), and the MCU does not produce anything in response.

[image]

I have a hypothesis that this may be related to the quartz frequency. The AN3154 has the following operating algorithm for the CAN bootloader (Figure 2):

[image]

It can be seen that they proceed from the assumption that the frequency of the external quartz is either 8 MHz, or 25 MHz, or 14.7456 MHz, and set the bitrate based on this. We have 20 MHz quartz in our mcu module... I can’t check this yet, because I don’t have quartz in the right package for these frequencies...

andreika-git, 2024-03-05:
Changing the crystal to 25 MHz didn't help... The firmware auto-detects it correctly and reports "25". The bootloader starts but no CAN activity... My next hypothesis is that the bootloader gets distracted by active USB connection. Maybe USB has higher priority than CAN. I'll try with 12V external power and no USB.

andreika-git, 2024-03-06:
It works!

[image]

The "USB hypothesis" is confirmed! When powered by an external 12V with no USB connected, then it's possible to connect via CAN.

rusefillc, 2024-07-09:
works on 125, 250, 500 or 1000 baud rate

```
  20:39:23 : Time elapsed during the read operation is: 00:00:00.092
  20:39:28 : Disconnected from device.
  20:39:38 : UR connection mode is defined with the HWrst reset mode
  20:39:41 : ST-LINK FW  : V3J8M1B5
  20:39:41 : Voltage     : 0.03V
  20:39:41 : Bridge Freq : 48000 KHz
  20:39:41 : Baudrate    : 1 Mbps
  20:39:41 : BL version  : 2.0
  20:39:41 : Device ID   : 0x419
  20:39:41 : UPLOADING OPTION BYTES DATA ...
  20:39:41 :   Bank          : 0x00
  20:39:41 :   Address       : 0x1fffc000
  20:39:41 :   Size          : 16 Bytes
  20:39:41 :   Bank          : 0x01
  20:39:41 :   Address       : 0x1ffec008
  20:39:41 :   Size          : 8 Bytes
  20:39:41 : UPLOADING ...
  20:39:41 :   Size          : 1024 Bytes
  20:39:41 :   Address       : 0x8000000
  20:39:41 : Read progress:
  20:39:41 : Data read successfully
  20:39:41 : Time elapsed during the read operation is: 00:00:00.094
```