rusefi/rusefi #5893: Enabling SPI-egt hangs the processor

zed65, 2024-01-31:
With: rusefi_bundle_f407-discovery 2024-01-30 17:47 a new problem have occurred.

When enabling SPI-EGT the processor hangs and the connection is lost.
Only solution is to flash the firmware again.

F407 Black board

dron0gus, 2024-02-02:
```
#0  chDbgPanic3 (msg=0x8072e70 "SV#4", file=0x8072e40 "ChibiOS/os/rt/src/chsys.c", line=201) at ./controllers/core/error_handling.cpp:108
#1  0x08002c28 in chSysHalt (reason=reason@entry=0x8072e70 "SV#4") at ChibiOS/os/rt/src/chsys.c:201
#2  0x08003a44 in _dbg_check_lock () at ChibiOS/os/rt/src/chdebug.c:151
#3  0x08004fdc in chSysLock () at ChibiOS/os/rt/include/chsys.h:359
#4  osalSysLock () at ChibiOS/os/hal/osal/rt-nil/osal.h:575
#5  spiStart (spip=0x20009660 <SPID3>, config=0x8072e70) at ChibiOS/os/hal/src/hal_spi.c:95
#6  0x08033d66 in readEgtPacket (egtChannel=egtChannel@entry=0) at ./hw_layer/sensors/max31855.cpp:91
#7  0x0806cb20 in getMax31855EgtValue (egtChannel=egtChannel@entry=0) at ./hw_layer/sensors/max31855.cpp:105
#8  0x08035a00 in grabEgtValues () at ./hw_layer/sensors/max31855.cpp:164
#9  0x0802b1fc in Engine::periodicSlowCallback (this=0x100056e8 <___engine>) at ./controllers/algo/engine.cpp:165
#10 0x0804f4a2 in doPeriodicSlowCallback () at ./controllers/engine_controller.cpp:198
#11 0x0807081c in PeriodicSlowController::PeriodicTask (this=<optimized out>) at ./controllers/engine_controller.cpp:136
#12 0x08033ec2 in runAndScheduleNext (controller=0x2001e988 <slowController>) at ./controllers/system/periodic_task.cpp:16
#13 0x08005042 in chVTDoTickI () at ChibiOS/os/rt/src/chvt.c:609
#14 0x08005070 in chSysTimerHandlerI () at ChibiOS/os/rt/src/chsys.c:347
#15 0x0805af16 in osalOsTimerHandlerI () at ChibiOS/os/hal/osal/rt-nil/osal.h:667
#16 st_lld_serve_interrupt () at ChibiOS/os/hal/ports/STM32/LLD/TIMv1/hal_st_lld.c:462
#17 0x0805af2c in SysTick_Handler () at ChibiOS/os/hal/ports/STM32/LLD/TIMv1/hal_st_lld.c:381
#18 <signal handler called>
#19 0x0805a952 in port_wait_for_interrupt () at ChibiOS/os/common/ports/ARMCMx/chcore_v7m.h:772
#20 _idle_thread (p=0x0) at ChibiOS/os/rt/src/chsys.c:79
#21 0x08000e0e in _port_thread_start () at ChibiOS/os/common/ports/ARMCMx/compilers/GCC/chcoreasm_v7m.S:201
```
SV#4. The function chSysLock() has been called from ISR context or from within a critical zone. This function is meant to start a critical zone from thread context. This can also happen when a normal API is called from within a critical zone.
https://www.chibios.org/dokuwiki/doku.php?id=chibios:documentation:books:rt:kernel_debug

Does PeriodicSlowController executed from ISR context?

rusefillc, 2024-02-03:
@zed65 please try with the fix (give it 20 minutes to build)

zed65, 2024-02-03:
Works perfect.

closed 2024-02-03 by zed65