rusefi/rusefi #5894: maximumIgnitionTiming does not acccount timing_offset_cylinder

rusefillc, 2024-01-31:
Set timing_offset_cylinder for a funny engine like Harley

problem: ignition is messed up due to default minimumIgnitionTiming and maximumIgnitionTiming no longer making sense
desired: same -10 to 60 IgnitionTiming range is still producing proper results on HD

We have maximumIgnitionTiming and minimumIgnitionTiming safety feature which does not account odd file engine shapes as it probably should

timing_offset_cylinder = "I have a funny engine with cylinders not evenly spaced every 720 / N degrees". Offsets both ignition and fueling.

mck1117, 2024-01-31:
Unless something is different on FOME, I'm not sure this is true. Clamping happens before adding the cylinder's offset, which includes both the offset in the firing order and the user-programmed cylinder offset.

rusefillc, 2024-01-31:
> Unless something is different

it's definitely different

rusefillc, 2025-07-04:
@FDSoftware was this resolved by https://github.com/rusefi/rusefi/pull/7957 or just relevant?

FDSoftware, 2025-07-04:
> [@FDSoftware](https://github.com/FDSoftware) was this resolved by [#7957](https://github.com/rusefi/rusefi/pull/7957) or just relevant?

Relevant (since we add the code need for this), not fixed yet, we need to modify a bit the ignition logic for solve this issue