rusefi/rusefi #5900: better solution without board name twice

rusefillc, 2024-02-01:
https://github.com/rusefi/rusefi/commit/5f2698ece3d1a0014b0cb912d3dae33d16905bd0#diff-7831f7c23817ec7253e6a585334c58982d07df0f584352ae1178884258055163

rusefillc, 2024-02-06:
> A draft for a PR:

@andreika-git please note SHORT_BOARD_NAME board.mk logic

andreika-git, 2024-02-06:
Also: meta-info.env 
Also: add _ARCH suffix to board_short_name for _some_ boards?

andreika-git, 2024-02-11:
https://github.com/rusefi/rusefi/issues/5931

andreika-git, 2024-02-11:
What's the deal with hellen154hyundai short board names? Why didn't it have hellen154hyundai_f4 and hellen154hyundai_f7?

rusefillc, 2024-02-12:
> What's the deal with hellen154hyundai's short board names? Why didn't it have hellen154hyundai_f4 and hellen154hyundai_f7?

something seems broken around compile_hellen154hyundai_f7.sh see https://github.com/rusefi/rusefi/commit/411f1c5bcb5db7d8fdf94ea8dfeabbadd593bf6f

rusefillc, 2024-02-12:
this one seems done next phase https://github.com/rusefi/rusefi/issues/5935

closed 2024-02-12 by rusefillc