rusefi/rusefi #5918: BT init error

rusefillc, 2024-02-05:
```
2024-02-05_18_07_11_215: EngineState: confirmation_bluetooth_jdy 115200 alphax 1234:32
2024-02-05_18_07_11_993: EngineState: *** Bluetooth module setup procedure ***
2024-02-05_18_07_14_979: EngineState: Restarting at 115200
2024-02-05_18_07_14_979: EngineState: sending AT  
2024-02-05_18_07_15_532: EngineState: Timeout waiting for BT reply
2024-02-05_18_07_15_533: EngineState: sending AT+DISC  
2024-02-05_18_07_15_990: EngineState: Timeout waiting for BT reply
2024-02-05_18_07_15_992: EngineState: Restarting at 9600
2024-02-05_18_07_15_994: EngineState: sending AT  
2024-02-05_18_07_16_089: EngineState: Received 5 +OK  
2024-02-05_18_07_16_090: EngineState: sending AT+VERSION  
2024-02-05_18_07_16_091: EngineState: Received 41 +VERSION=JDY-33-V1.1,Bluetooth V3.0+BLE  
2024-02-05_18_07_16_093: EngineState: sending AT+BAUD  
2024-02-05_18_07_16_093: EngineState: Received 9 +BAUD=4  
2024-02-05_18_07_16_094: EngineState: sending AT+TYPE  
2024-02-05_18_07_16_202: EngineState: Received 9 +TYPE=1  
2024-02-05_18_07_16_205: EngineState: sending AT+PIN  
2024-02-05_18_07_16_206: EngineState: Received 11 +PIN=1234  
2024-02-05_18_07_16_207: EngineState: sending AT+LADDR  
2024-02-05_18_07_16_209: EngineState: Received 21 +LADDR=7D12CF4AE614  
2024-02-05_18_07_16_211: EngineState: sending AT+STAT  
2024-02-05_18_07_16_211: EngineState: Received 10 +STAT=00  
2024-02-05_18_07_16_214: EngineState: sending AT+DEFAULT  
2024-02-05_18_07_16_313: EngineState: Received 5 +OK  
2024-02-05_18_07_16_314: EngineState: sending AT+ENLOG0  
2024-02-05_18_07_16_315: EngineState: Received 5 +OK  
2024-02-05_18_07_16_316: EngineState: sending AT+TYPE0  
2024-02-05_18_07_16_318: EngineState: Received 5 +OK  
2024-02-05_18_07_16_319: EngineState: sending AT+BAUD8  
2024-02-05_18_07_16_422: EngineState: Received 5 +OK  
2024-02-05_18_07_16_423: EngineState: sending AT+NAMEalphax  
2024-02-05_18_07_16_862: EngineState: Timeout waiting for BT reply
2024-02-05_18_07_16_862: EngineState: FAIL! Command AT+NAMEalphax   failed
2024-02-05_18_07_35_862: EngineState: confirmation_bluetooth_jdy 115200 alphax 1234:32
2024-02-05_18_07_35_862: EngineState: *** Bluetooth module setup procedure ***
2024-02-05_18_07_38_929: EngineState: Restarting at 115200
2024-02-05_18_07_38_930: EngineState: sending AT  
2024-02-05_18_07_38_930: EngineState: Received 5 +OK  
2024-02-05_18_07_38_931: EngineState: sending AT+VERSION  
2024-02-05_18_07_38_931: EngineState: Received 41 +VERSION=JDY-33-V1.1,Bluetooth V3.0+BLE  
2024-02-05_18_07_38_931: EngineState: sending AT+BAUD  
2024-02-05_18_07_38_933: EngineState: Received 9 +BAUD=8  
2024-02-05_18_07_38_933: EngineState: sending AT+TYPE  
2024-02-05_18_07_38_940: EngineState: Received 9 +TYPE=0  
2024-02-05_18_07_38_941: EngineState: sending AT+PIN  
2024-02-05_18_07_38_941: EngineState: Received 11 +PIN=1234  
2024-02-05_18_07_38_943: EngineState: sending AT+LADDR  
2024-02-05_18_07_38_944: EngineState: Received 21 +LADDR=7D12CF4AE614  
2024-02-05_18_07_38_947: EngineState: sending AT+STAT  
2024-02-05_18_07_38_947: EngineState: Received 10 +STAT=00  
2024-02-05_18_07_38_949: EngineState: sending AT+DEFAULT  
2024-02-05_18_07_38_949: EngineState: Received 5 +OK  
2024-02-05_18_07_38_951: EngineState: sending AT+ENLOG0  
2024-02-05_18_07_39_039: EngineState: Too long reply from BT
2024-02-05_18_07_39_039: EngineState: sending AT+TYPE0  
2024-02-05_18_07_39_488: EngineState: Timeout waiting for BT reply
2024-02-05_18_07_39_489: EngineState: sending AT+BAUD8  
2024-02-05_18_07_40_034: EngineState: Timeout waiting for BT reply
2024-02-05_18_07_40_035: EngineState: FAIL! Command AT+BAUD8   failed

```

rusefillc, 2024-02-09:
ops that JDY-33 looks different from 4chan?!

[image]

rusefillc, 2024-02-10:
@dron0gus says he has a module like on the blue 4chan but unlike the piece I have his has chip marking removed as well?!

[image]

rusefillc, 2024-03-10:
Same thing with uaEFI which are pre-populated with JDY-33

rusefillc, 2024-03-10:
the less broken JDY-33 I have in NJ

[image]

rusefillc, 2024-03-10:
```
2024-03-10_19_18_55_886: EngineState: confirmation_bluetooth_bk 115200 alphax 1234:31
2024-03-10_19_18_56_093: EngineState: *** Bluetooth module setup procedure ***
2024-03-10_19_18_56_093: EngineState: !Warning! Please make sure you're not currently using the BT module for communication (not paired)!
2024-03-10_19_18_56_094: EngineState: TO START THE PROCEDURE: PLEASE DISCONNECT YOUR PC COM-PORT FROM THE BOARD NOW!
2024-03-10_19_18_56_094: EngineState: After that please don't turn off the board power and wait for ~15 seconds to complete. Then reconnect to the board!
2024-03-10_19_18_59_129: EngineState: Restarting at 115200
2024-03-10_19_18_59_130: EngineState: sending AT  
2024-03-10_19_18_59_627: EngineState: Timeout waiting for BT reply
2024-03-10_19_18_59_627: EngineState: Restarting at 9600
2024-03-10_19_18_59_704: EngineState: sending AT  
2024-03-10_19_18_59_705: EngineState: Received 5 +OK  
2024-03-10_19_18_59_705: EngineState: sending AT+BAUD8  
2024-03-10_19_18_59_706: EngineState: Received 5 +OK  
2024-03-10_19_18_59_708: EngineState: sending AT+NAMEalphax  
2024-03-10_19_19_00_231: EngineState: Timeout waiting for BT reply
2024-03-10_19_19_00_231: EngineState: FAIL! Command AT+NAMEalphax   failed
```

rusefillc, 2024-03-10:
```
2024-03-10_19_22_14_866: EngineState: confirmation_bluetooth_bk 115200 alphax 1234:31
2024-03-10_19_22_15_173: EngineState: *** Bluetooth module setup procedure ***
2024-03-10_19_22_15_174: EngineState: !Warning! Please make sure you're not currently using the BT module for communication (not paired)!
2024-03-10_19_22_15_174: EngineState: TO START THE PROCEDURE: PLEASE DISCONNECT YOUR PC COM-PORT FROM THE BOARD NOW!
2024-03-10_19_22_15_174: EngineState: After that please don't turn off the board power and wait for ~15 seconds to complete. Then reconnect to the board!
2024-03-10_19_22_18_189: EngineState: Restarting at 115200
2024-03-10_19_22_18_189: EngineState: sending AT  
2024-03-10_19_22_18_282: EngineState: Received 5 +OK  
2024-03-10_19_22_18_282: EngineState: sending AT+BAUD8  
2024-03-10_19_22_18_282: EngineState: Received 5 +OK  
2024-03-10_19_22_18_282: EngineState: sending AT+NAMEalphax  
2024-03-10_19_22_18_282: EngineState: Received 5 +OK  
2024-03-10_19_22_18_283: EngineState: sending AT+PIN1234  
2024-03-10_19_22_18_286: EngineState: Received 5 +OK  
2024-03-10_19_22_18_287: EngineState: SUCCESS! All commands passed to the Bluetooth module!
2024-03-10_19_22_18_290: EngineState: UART: TS: ERROR: no second byte
```

rusefillc, 2024-03-11:
Wow, it actually functions - it just does not want to change BT name?!

[image]

rusefillc, 2024-04-03:
[Saleh Mohammad](https://www.facebook.com/groups/938729756561668/user/100053332039474/?__cft__[0]=AZVY6J_wuCToRzUag6K8XSyKuoPUVudLjAbjJwQTDPoPufW-_bdkb2hcG-QzwPGMkm_2iwelp80IutpqK-B0bxJGOlrrbaIqqy77Yj70gIOMWGhMmlbs6AF1zC9vPS0E-3J1b5OyF5WFILIxUadErHPv&__tn__=R]-R)
Author
Top Contributor
[Andrey Be](https://www.facebook.com/groups/938729756561668/user/1420214202/?__cft__[0]=AZVY6J_wuCToRzUag6K8XSyKuoPUVudLjAbjJwQTDPoPufW-_bdkb2hcG-QzwPGMkm_2iwelp80IutpqK-B0bxJGOlrrbaIqqy77Yj70gIOMWGhMmlbs6AF1zC9vPS0E-3J1b5OyF5WFILIxUadErHPv&__tn__=R]-R) I found why the name change fails , you should change the order of AT commands and let baud change become last command because the baud change will break the sequence and the command will fail , all of those commands can be done without interference with the serial connection however baudrate needs at least some sort of halt command switch rate then continue however change of sequence is better keeping baud the last command will fix all issues!

mck1117, 2024-04-24:
I have some jdy33 with both pcb layouts, and whether the chip is lasered clean or not, the ones matching the pic in the first post don't work, while the ones with the other layout do work

rusefillc, 2024-05-02:
fake is probably the wrong term, but something funny for sure?
JDY is not using a very reputable chip in the first place?

rusefillc, 2024-06-26:
my current state https://rusefi.com/forum/viewtopic.php?p=49666#p49666

rusefillc, 2024-12-03:
See also https://rusefi.com/forum/viewtopic.php?f=4&t=2854

exvils, 2024-12-17:
`bluetooth_jdy 115200 JDY-33-SPP 1234`
passes without problems on aliexpress jdy33 (1,50€ board). Thing is keep stock name as it wont allow change via uaefi.