rusefi/rusefi #5924: existing FatFS logs to support SPI flash

rusefillc, 2024-02-08:
* we access SD via USB
* SD card inserted into a holder is just weird if we want to seal a case forever

thus we shall support logging on SPI flash. Would running FatFS on an SPI flash be the simplest way to achieve?

mck1117, 2024-02-08:
JFYI the price per GB of SPI flash is extremely bad compared to microSD cards. It's cheaper (and often better- you get a full NAND controller with wear leveling for free!) to just glue down a micro sd card.

rusefillc, 2024-02-08:
> JFYI the price per GB of SPI flash is extremely bad compared to microSD cards. It's cheaper (and often better- you get a full NAND controller with wear leveling for free!) to just glue down a micro sd card.

Yes, engineering-wise you are 200% correct. But humans just do not perceive hot glue well.

Great news https://www.eevblog.com/forum/projects/solderable-sd-card/ points at https://www.adafruit.com/product/4899 https://cdn-shop.adafruit.com/product-files/4899/2005251034_XTX-XTSD04GLGEAG_C558839%282%29.pdf which is discontinued but same brand seems to have newer products https://www.lcsc.com/brand-detail/12264.html

mck1117, 2024-02-08:
you're looking for https://en.wikipedia.org/wiki/MultiMediaCard#eMMC

zed65, 2024-02-08:
I'm about to test a XTSD04GLGEAG on my new board.

rusefillc, 2024-02-09:
> The tests I've done so far works with XTSD04GLGEAG

how did you FatFS format it?

zed65, 2024-02-09:
I didn't. I bought this https://www.adafruit.com/product/4899 and desoldered the chip. Maybe it was already formated.

rusefillc, 2024-07-12:
CSNP1GCR01-BOW
http://www.longsto.com/product/30.html

https://jlcpcb.com/partdetail/Creat_StorageWorld-CSNP1GCR01BOW/C2691593 LGA-8(6x8)

https://www.lcsc.com/datasheet/lcsc_datasheet_2102221705_Creat-Storage-World-CSNP1GCR01-BOW_C2691593.pdf mentions SPI mode

rusefillc, 2025-08-20:
> you're looking for https://en.wikipedia.org/wiki/MultiMediaCard#eMMC

8-bit parallel interface? :(