rusefi/rusefi #5925: Expand daihatsu 3+1 3cyl trigger to 4+1 as well

runesmdk, 2024-02-08:
Daihatsu uses the n+1 across both 3 and 4 cyl. 
The wiki describes daihatsu trigger as only 3+1 now. 

Their 4+1 works exactly at the same +1 timing as the 3+1, its 4 pulses every 720° crank, and then a single extra pulse at 30° crank.

rusefillc, 2024-02-08:
@runesmdk see https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#daihatsu

runesmdk, 2024-02-08:
> @runesmdk see https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#daihatsu

That went fast! 

The +1 signal you are referring to isnt tdc however, the tdc is the one before the +1. So the signals fall 0, 30, 180, 360, 540 (atleast according to speeduino wikipedia).
I dont know if that means anything for the rusefi or it just tracks crank degrees. 

Also, the rusefi wiki about triggers dont mention it being coded as a cam-speed signal, i just assume it is?

runesmdk, 2024-02-08:
"Syncedge rise only", if i use a hall sensor, the signal falls when the sensor signal goes low.

rusefillc, 2024-02-08:
@runesmdk updated https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#daihatsu

closed 2024-02-08 by rusefillc

runesmdk, 2024-02-08:
> @runesmdk updated https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#daihatsu

Bang on. What about my previous message about rising only?

rusefillc, 2024-02-08:
> What about my previous message about rising only?

let's think about it once your engine is idling with current firmware and we see any problems

runesmdk, 2024-02-08:
> > What about my previous message about rising only?
> 
> let's think about it once your engine is idling with current firmware and we see any problems

The windows on the shutter plate for the hall sensor only fits with falling as the 180° signal shutters arent equal length on the rising, especially not on the +1

rusefillc, 2024-02-08:
reminder that edge is configurable

[image]

runesmdk, 2024-02-08:
> reminder that edge is configurable [...]

[image]
[image]

Okay thats great. I was only asking because i saw that