rusefi/rusefi #5935: build scripts monoliht

rusefillc, 2024-02-12:
* openblt compilation wipes out incremental main firmware state due to pch coupling. in order to have incremental compilation we shall decouple
* autoupdate bundle and full bundle are weirdly coupled
* it's unclear what are the key inputs: is it folder name or shortBoardId or else?

chuckwagoncomputing, 2024-02-12:
Please be aware of my current effort to move all build scripts to Make, which I believe will solve this issue.

rusefillc, 2024-02-12:
> Please be aware of my current effort to move all build scripts to Make, which I believe will solve this issue.

@andreika-git @chuckwagoncomputing how to coordinate this better? @chuckwagoncomputing can you focus for instance on incremental build with openblt and deal with gen_live_data later, or is all that not easily divisible?

chuckwagoncomputing, 2024-02-13:
OK I'll try to get a PR-able state that fixes incremental builds.
Current state of the project: https://github.com/rusefi/rusefi/compare/master...chuckwagoncomputing:rusefi:makeover?expand=1

chuckwagoncomputing, 2024-02-13:
Part 0 https://github.com/rusefi/rusefi/pull/5936

rusefillc, 2024-02-13:
@chuckwagoncomputing what's your proposal for meta info discovery? i.e. what's the unique key and how is everything else resolved from it?

corner cases are mre_f4 and mre_f7 while 4chan and 4chan_f7

chuckwagoncomputing, 2024-02-13:
Two keys:
- board directory
- build target

Everything else is contained at <board directory>/compile_<build target>.sh

rusefillc, 2024-02-13:
PS: @chuckwagoncomputing another use-case is https://github.com/rusefi/rusefi/wiki/Custom-Firmware

specifically desire is to have "paralela" mentioned only once within the whole 'https://github.com/rusefi/fw-custom-paralela' repo. Test criteria is "how many entities need renaming/editing once https://github.com/rusefi/fw-custom-example is forked"

since technically at the moment we have one key in compile_XXX.sh file name but there is also compile_proteus_f4_debug.sh which is proteus_f4 but with debug

rusefillc, 2024-02-13:
PPS @chuckwagoncomputing https://github.com/rusefi/fw-custom-paralela/blob/master/.github/workflows/build-firmware.yaml is an interesting case of meta info consumption

chuckwagoncomputing, 2024-02-13:
Yes I'm working on that before making https://github.com/rusefi/rusefi/pull/5938 not a draft. I didn't know about the desire to have the name mentioned only once though; that's a wrench in the gears.

rusefillc, 2024-02-13:
>  I didn't know about the desire to have the name mentioned only once though; that's a wrench in the gears.

I mean nothing is set in stone but just to make it less error-prone and hopefully common sense says that if we fork https://github.com/rusefi/fw-custom-example we produce _one_ entity which ideally should have _one_ key mentioned _once_?

chuckwagoncomputing, 2024-02-18:
Interesting note regarding keys:
We are now using board directory and bundle name as keys, and then resolving them to a meta-info file. We could use only the meta-info file.
- Get BUNDLE_NAME from the meta-info file name
- Assume BOARD_DIR is the directory the file is in

chuckwagoncomputing, 2024-02-18:
That's correct. And the BOARD_DIR for all those files is the directory they are in.

I didn't think of it before, but getting BUNDLE_NAME from the meta-info file will require renaming all of the default 'meta-info.env' files to include the bundle name. Or we could just default to using SHORT_BOARD_NAME if the file is meta-info.env.

chuckwagoncomputing, 2024-02-18:
Actually [common_script_read_meta_info.inc](https://github.com/rusefi/rusefi/blob/master/firmware/config/boards/common_script_read_meta_env.inc) already sets BOARD_DIR to the directory the meta-info file is in.

chuckwagoncomputing, 2024-03-08:
I'm going to call this completed.

closed 2024-03-08 by chuckwagoncomputing