rusefi/rusefi #5952: ETB traction control drop is affected by not-highlighted cell

ElDominio, 2024-02-16:
**Describe the bug**
If current slip state is 1.00 and i have two cells in Y column, 1.02 and 1.03., if I write values in 1.03 ETB drops anyways

**Hardware**
 - hellen154hyundai

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Controller ->Traction Control ETB Drop
2. write vlues in a non-highlighted cell
3. ETB drops

**Expected behavior**
Does not drop ETB in cells not close to the ETB drop

**Screenshots**
https://youtu.be/WwZlngGoSQc

zed65, 2024-02-16:
Related to:??
[image]

rusefillc, 2024-03-04:
Could be consequence of https://github.com/rusefi/rusefi/issues/6139

commit 4b6786b6 (2024-03-04): ETB traction control drop is affected by not-highlighted cell fix #5952
closed 2024-03-04 by rusefillc