rusefi/rusefi #5970: RFC: What to do with compile_.sh scripts

chuckwagoncomputing, 2024-02-17:
As part of my current "makeover" project, the build-firmware workflow will call `make` directly, not using the `compile_<build name>.sh` scripts found in each board directory.

I'm not sure what I should do with these scripts.
- delete them
- leave them for humans to use

If I leave them for humans to use, there are a few options on how they should work:
- source the meta-info file using common_script_read_meta_env and call make
- call [the new compile script](https://github.com/rusefi/rusefi/blob/master/firmware/bin/compile.sh) passing the meta-info file

rusefillc, 2024-02-17:
Q1: definitely leave them for humans
Q2: whatever takes less time right now since good enough would work as long as it works?

chuckwagoncomputing, 2024-02-17:
I'm leaning towards calling the new compile script so that in case we want to add something like `make clean` or `gen_config_board` we can do it in one place.

PR #5971 (2024-02-17): Makeover compile call
Also moved some declarations in frankenso-na6 I missed earlier because they weren't `export`ed

fix #5970
closed 2024-02-17 by rusefillc