rusefi/rusefi #5980: internal MAP not selectable as input in UAEFI TS

ggurov, 2024-02-18:
**Describe the bug**
Internal map input channel is not selectable for any sensor

**Hardware**
 - Board: uaefi

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Baro sensor'
2. Click on 'Baro input'
3. Scroll down to where the list ends
4. Not see internal map sensor as input

**Expected behavior**
Ability to select PC1 pin as input for sensors

**Additional context**
Schematics say IN_MAP2_ PC1, next to PC0 coming from molex MAP

rusefillc, 2024-02-18:
@ggurov thank you for the defect report! Should be better now https://github.com/rusefi/rusefi/commit/2bcc7d37e27a780acafe55c91faf3cca6134bf20

closed 2024-02-18 by rusefillc

ggurov, 2024-02-18:
[image]
 
LGTM