rusefi/rusefi #5986: do we run all GHA _twice_ per pull request?

rusefillc, 2024-02-18:
See https://github.com/rusefi/rusefi/pull/5977

We have both a failed run and a happy run?! is that one run for "pull_request" and one run for "push"? is that a bug or a feature? shall we simply remove "pull_request" from "on" in all yaml?

[image]

chuckwagoncomputing, 2024-02-18:
I think it's because it's a branch in the same repo.
It's running once on the branch, and once in the pull request.
We still want pull-request checks.

rusefillc, 2024-02-18:
ohhh so it's not _any_ PR it's my own PRs?

One day not urgently I think we need to check that corner case and quick-pass in order to reduce GHA queuing.

rusefillc, 2024-02-18:
PS: unless we improve this to get green https://github.com/rusefi/rusefi/pull/5977

but that would not automatically give us green master :(