rusefi/rusefi #5989: Where did VR threshold config go?

ElDominio, 2024-02-18:
**Describe the bug**
VR Threshold settings vanished? Need to configure low rpm threshold 

**Hardware**
 - Board: 4chan
 - Engine info: not real engine

**To Reproduce**
Steps to reproduce the behavior:
1. Look for VR Threshold
2. Cannot find

**Expected behavior**
why was this configurability removed?

rusefillc, 2024-02-18:
"look" sounds like something a human would do? Proper robotic way is to use "search" feature of TS

Screen shot shows snapshot version and search results. I do not see an issue please clarify @ElDominio 

[image]

ElDominio, 2024-02-18:
[image]

Fw v20240120

ElDominio, 2024-02-18:
Was removed at some point apparently but I updated and it's back

closed 2024-02-18 by ElDominio