rusefi/rusefi #6002: proteus F7 HW CI is not 100% reliable

rusefillc, 2024-02-20:
NJ F7 proteus HW CI setup sometimes fails on openocd phase

full manual power cycle of Proteus F7 resolves this issue

```
Info : clock speed 2000 kHz
Info : STLINK V3J3M2 (API v3) VID:PID 0483:374E
Info : Target voltage: 3.309188
Info : PROTEUS.cpu: hardware has 8 breakpoints, 4 watchpoints
Info : starting gdb server for PROTEUS.cpu on 3333
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* PROTEUS.cpu        hla_target little PROTEUS.cpu        running

Info : Unable to match requested speed 2000 kHz, using 1000 kHz
Info : Unable to match requested speed 2000 kHz, using 1000 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x002003d8 msp: 0x20021000
Info : device id = 0x10016451
Info : flash size = 1024 kbytes
Info : Dual Bank 1024 kiB STM32F76x/77x found
erased sectors 0 through [15](https://github.com/rusefi/rusefi/actions/runs/7965708559/job/21745745392#step:10:16) on flash bank 0 in [17](https://github.com/rusefi/rusefi/actions/runs/7965708559/job/21745745392#step:10:18).821079s

wrote 46[18](https://github.com/rusefi/rusefi/actions/runs/7965708559/job/21745745392#step:10:19)00 bytes from file build/rusefi.bin in 5.888227s (76.590 KiB/s)

Info : Unable to match requested speed [20](https://github.com/rusefi/rusefi/actions/runs/7965708559/job/21745745392#step:10:21)00 kHz, using 1000 kHz
Info : Unable to match requested speed 2000 kHz, using 1000 kHz

Error: Process completed with exit code 1.
```

rusefillc, 2024-02-20:
[image]

chuckwagoncomputing, 2024-03-24:
Have you found any way to trigger this? I've not been able to reproduce it by running openocd_wipe_and_flash.sh repeatedly.

chuckwagoncomputing, 2024-03-29:
I just ran 100 back-to-back build-flash-test cycles with success.