rusefi/rusefi #6008: HARDWARE VALIDATION FAILED error on multiple units

ElDominio, 2024-02-20:
**Describe the bug**
An odd critical error has been affecting several units, not letting them run

**Hardware**
 - Board:8chan revC, F407 paralela
 - Engine infoford coyote, 4G93

**To Reproduce**
Steps to reproduce the behavior:
1. Connect usb without 12v
2. See HARDWARE VALIDATION FAILED error

**Expected behavior**
This error does not exist? The error is quite cryptic

**Screenshots**
[image]

**Additional context**

closed 2024-02-20 by rusefillc

ElDominio, 2024-02-21:
But this was on F407, not 8chan?

rusefillc, 2024-02-21:
@ElDominio ops. I've missed that part. Just reverted 8chan change

@ElDominio please consider making such important details more obvious by placing into the ticket title instead of vague "multiple units".

rusefillc, 2024-02-21:
> But this was on F407, not 8chan?

C'mon. Please do much better than "F407".

I refuse to understand "F407" as a reference to any board.

rusefillc, 2024-02-21:
@ElDominio https://github.com/rusefi/fw-custom-paralela/commit/450b5312bb8bd1019ef3178faaea2992f0b5feff

commit 51eb64d1 (2024-02-20): HARDWARE VALIDATION FAILED error on multiple 8chan units fix #6008
commit 9f9ba88f (2024-02-21): Revert "HARDWARE VALIDATION FAILED error on multiple 8chan units fix #6008"

This reverts commit 51eb64d1191e75ccc256efbc30c2113dd54aa856.
commit 71acf2ac (2024-02-21): Revert "HARDWARE VALIDATION FAILED error on multiple 8chan units fix #6008"

This reverts commit 51eb64d1191e75ccc256efbc30c2113dd54aa856.