rusefi/rusefi #6014: bundle.mk:127: warning: overriding recipe for target

rusefillc, 2024-02-21:
```
rusefi@DESKTOP-57S88S8:~/rusefi/firmware$ make clean
PROJECT_BOARD: f407-discovery
BOARD_DIR:     ././config/boards/f407-discovery
PROJECT_CPU:   ARCH_STM32F4
SHORT_BOARD_NAME: f407-discovery
CPU_HWLAYER:   ports/stm32/stm32f4
CONFDIR:       ./hw_layer/ports/stm32/stm32f4/cfg
LDSCRIPT:      ./hw_layer/ports/stm32/stm32f4/STM32F4.ld
bundle.mk:127: warning: overriding recipe for target '&'
bundle.mk:107: warning: ignoring old recipe for target '&'
```

rusefillc, 2024-02-22:
```
rusefi@DESKTOP-57S88S8:~/rusefi/firmware$ git log
commit e04afc1cf952ce8174109f980283bf66cd31d331 (HEAD -> master, origin/master, origin/HEAD)
Author: David Holdeman <dave@holdemanenterprises.com>
Date:   Wed Feb 21 18:01:07 2024 -0600

    Conditionally set bootloader output for prerequisite only

commit 461426bfd59d183fb812accdbeafa38228c99a21
Author: GitHub gen-default-tune Action <action@github.com>
Date:   Thu Feb 22 01:24:59 2024 +0000

    Auto-generated default tune

commit 76b1bbe3ba184f6c1be4feea48dda90d338126c2
```

```
rusefi@DESKTOP-57S88S8:~/rusefi/firmware$ make clean
PROJECT_BOARD: f407-discovery
BOARD_DIR:     ././config/boards/f407-discovery
PROJECT_CPU:   ARCH_STM32F4
SHORT_BOARD_NAME: f407-discovery
CPU_HWLAYER:   ports/stm32/stm32f4
CONFDIR:       ./hw_layer/ports/stm32/stm32f4/cfg
LDSCRIPT:      ./hw_layer/ports/stm32/stm32f4/STM32F4.ld
bundle.mk:132: warning: overriding recipe for target '&'
bundle.mk:112: warning: ignoring old recipe for target '&'
Cleaning PCH
rm -f ./pch/pch.h.gch/firmware
Cleaning Bundle
make -C ../simulator clean
make[1]: Entering directory '/home/rusefi/rusefi/simulator'
OS is []
```

chuckwagoncomputing, 2024-02-22:
Can you post the output of `make -v`?

rusefillc, 2024-02-22:
```
rusefi@DESKTOP-57S88S8:~/rusefi/firmware$ make -v
GNU Make 4.2.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
rusefi@DESKTOP-57S88S8:~/rusefi/firmware$ uname -a
Linux DESKTOP-57S88S8 5.15.133.1-microsoft-standard-WSL2 #1 SMP Thu Oct 5 21:02:42 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
rusefi@DESKTOP-57S88S8:~/rusefi/firmware$
```

chuckwagoncomputing, 2024-02-22:
Your make is 7+ years out of date and doesn't support grouped targets. I'd tell you to upgrade, but you won't be the last with the same problem, so we should probably do a backwards-compatible workaround instead.

chuckwagoncomputing, 2024-02-22:
In particular, Mac users will have issues because the make that comes preinstalled is really old. The could install a newer version from homebrew or something, but it's easier to work around it than to try to educate everyone.

rusefillc, 2024-02-22:
> it's easier to work around it than to try to educate everyone.

a very wide statement. See Fahrenheit :)

chuckwagoncomputing, 2024-02-22:
Is this fixed for you?

rusefillc, 2024-02-22:
yes!

```
rusefi@DESKTOP-57S88S8:~/rusefi/firmware$ make clean
PROJECT_BOARD: f407-discovery
BOARD_DIR:     ././config/boards/f407-discovery
PROJECT_CPU:   ARCH_STM32F4
SHORT_BOARD_NAME: f407-discovery
CPU_HWLAYER:   ports/stm32/stm32f4
CONFDIR:       ./hw_layer/ports/stm32/stm32f4/cfg
LDSCRIPT:      ./hw_layer/ports/stm32/stm32f4/STM32F4.ld
Cleaning PCH
rm -f ./pch/pch.h.gch/firmware
Cleaning Bundle
make -C ../simulator clean
make[1]: Entering directory '/home/rusefi/rusefi/simulator'
OS is []
Cleaning Simulator
rm -rf build
rm -rf .dep
Done Cleaning Simulator
Cleaning
- .dep
- build

Done
make[1]: Leaving directory '/home/rusefi/rusefi/simulator'
BOARD_DIR=../././config/boards/f407-discovery BOARD_META_PATH=../ make -C bootloader clean
make[1]: Entering directory '/home/rusefi/rusefi/firmware/bootloader'
Cleaning PCH
rm -f ../pch/pch.h.gch/bootloader
Cleaning-BLT
rm -fR blbuild/*
Cleaning-GCC
- .dep
- blbuild

Done
make[1]: Leaving directory '/home/rusefi/rusefi/firmware/bootloader'
rm -rf rusefi.snapshot.f407-discovery
Done Cleaning Bundle
Cleaning-GCC
- .dep
- build

Done
rusefi@DESKTOP-57S88S8:~/rusefi/firmware$
```

chuckwagoncomputing, 2024-02-22:
Are you having this issue again? If so, could you post output?

rusefillc, 2024-02-22:
Nope, that was me considering a rollback to fix master

PR #6019 (2024-02-22): Conditionally set bootloader output for prerequisite only
Should fix #6014
closed 2024-02-22 by rusefillc